geerlingguy/raspberry-pi-dramble #7: Consider using external USB SSD for MySQL server

geerlingguy, 2015-02-15:
I have an old 64GB USB 3.0 SSD I pulled from an old MacBook Air, and it gets 150-300MB/sec transfer, with quite low latency. It would definitely saturate the USB 2.0 bus on the Pi, but it might be a heck of a lot faster as the data store for MySQL than using the normal mount inside the cheap Kingston microSD card!

It's worth investigating at least, and maybe in the database playbook, the configuration could allow for internal vs. external drive configuration.

geerlingguy, 2015-02-15:
Just for comparison, doing the same `dd` test on my little MacBook Air, I got `620.4 MB/s`. Someday the Pi will have SATA at least, and/or USB 3.0. Someday...

closed 2015-02-18 by geerlingguy

geerlingguy, 2015-02-19:
Using the same Transcend 300x card as with the B+, I re-ran all the tests on a new Raspberry Pi 2 model B:

## `hdparm` - raw read speeds

(Prerequisite: `sudo apt-get install -y hdparm`).

```
$ sudo hdparm -t /dev/mmcblk0
 Timing buffered disk reads:  54 MB in  3.07 seconds =  17.58 MB/sec

$ sudo hdparm -T /dev/mmcblk0
 Timing cached reads:   866 MB in  2.00 seconds = 432.48 MB/sec
```

## `dd` - raw write speeds

```
$ sudo dd if=/dev/zero of=/tmp/output bs=8k count=25k; sudo rm -f /tmp/output
209715200 bytes (210 MB) copied, 4.2601 s, 49.2 MB/s
```

## `iozone` - practical read/write scenarios

```
$ ./iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4     1593     1594     5752     5785     5507      991
102400     512     8143    12418    19002    18770    18919     6963
102400   16384    13021    13030    19191    19190    19093    10729
```

geerlingguy, 2015-02-19:
## SanDisk Extreme Pro 8GB card on Raspberry Pi 2

Prep instructions:

  1. Boot Pi
  2. Expand filesystem with `sudo raspi-config` (might need to `sudo swapoff --all` prior to this, and delete `/var/swap` file)
  3. Reboot Pi
  4. Install hdparm (`sudo apt-get install -y hdparm`)
  5. Install ioperf
  6. Run tests

### `hdparm` - raw read speeds

```
$ sudo hdparm -t /dev/mmcblk0
TODO

$ sudo hdparm -T /dev/mmcblk0
TODO
```

### `dd` - raw write speeds

```
$ sudo dd if=/dev/zero of=/tmp/output bs=8k count=25k; sudo rm -f /tmp/output
TODO
```

### `iozone` - practical read/write scenarios

```
$ ./iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2
TODO
```

geerlingguy, 2015-02-19:
## SanDisk Extreme 32GB card on Raspberry Pi 2

### `hdparm` - raw read speeds

```
$ sudo hdparm -t /dev/mmcblk0
TODO

$ sudo hdparm -T /dev/mmcblk0
TODO
```

### `dd` - raw write speeds

```
$ sudo dd if=/dev/zero of=/tmp/output bs=8k count=25k; sudo rm -f /tmp/output
TODO
```

### `iozone` - practical read/write scenarios

```
$ ./iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2
TODO
```

geerlingguy, 2015-02-19:
## Samsung 16GB EVO card on Raspberry Pi 2

### `hdparm` - raw read speeds

```
$ sudo hdparm -t /dev/mmcblk0
TODO

$ sudo hdparm -T /dev/mmcblk0
TODO
```

### `dd` - raw write speeds

```
$ sudo dd if=/dev/zero of=/tmp/output bs=8k count=10k; sudo rm -f /tmp/output
TODO
```

### `iozone` - practical read/write scenarios

```
$ ./iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /home/pi/test
TODO
```

geerlingguy, 2015-02-20:
...and here are some rather depressing read/write stats (depressing because they show how much faster a proper USB 3.0 bus is for these fancy cards) from my MacBook Air using a USB 3.0 microSD card reader:

**SanDisk Extreme Pro 8GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4    80292    63909  1926436  1938478  1748930    80047
102400     512    77696    78564  1540404  1662500  1477503    80289
102400   16384    78283    76706  1354555  1458234  1181668    79245
```

**SanDisk Extreme 16GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4    43757    52241  1894611  2015954  1582739    52435
102400     512    52109    52293  1903132  1783470  1502830    52304
102400   16384    51984    52338  1730854  1748188  1573843    52300
```

**SanDisk Ultra 16GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4    11442     9014  2246112  2270165  1876278     9913
102400     512     9974     9850  1744433  1884916  1653208    10260
102400   16384     9665    10601  1672116  1864254  1473050    10523
```

**Samsung EVO 16GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4    10998     9033  1893976  1899782  1661497    10372
102400     512     9669    11573  1825992  1621869  1501863    12064
102400   16384    12692    12402  1784938  1547510  1422219    12280
```

**Transcend Premium 300x 32GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4     6551     8203  2127877  2136270  1831122     9094
102400     512    10806    12816  1991482  1869166  1704590    14685
102400   16384    13404    14460  1711214  1870742  1399345    15028
```

**Kingston Class 10 8GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4     7066     6885  1828370  1639189  1482317     8684
102400     512    10381    10020  1620914  1564789  1660931    10023
102400   16384    10424    10101  1665013  1522791  1445413    10422
```

**Cheapo Class 4 4 GB**

```
$ iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2 -f /Volumes/Test/test
                                                    random    random
    kB  reclen    write  rewrite    read    reread    read     write
102400       4     8865     7924  2168936  2119958  1860704     7940
102400     512     7404     5932    13780    14530    14287     7147
102400   16384     8066     5887  1718002  1730338  1516635     5998
```

geerlingguy, 2015-02-28:
I've compiled these benchmarks here:

https://github.com/geerlingguy/raspberry-pi-dramble/wiki/microSD-Card-Benchmarks

geerlingguy, 2015-02-28:
I've also added some info about external USB drives here:

https://github.com/geerlingguy/raspberry-pi-dramble/wiki/External-USB-drives

mikkorantalainen, 2017-12-02:
If you really want to test flash memory performance, you need to use `fio`. Using `dd` or some else software that is *not designed* for this purpose is not a valid replacement.

commit a01a82a0 (2015-02-18): Fixes #7: Storage IO benchmarks.