geerlingguy/raspberry-pi-dramble #51: Compile PHP 5.5 or 5.6, since only 5.4 available in repos

geerlingguy, 2015-07-04:
Raspberry Pi-compatible (ARM) apt repos only contain PHP versions up to 5.4.x. A couple options:

  - Compile PHP from source (supported easily by `geerlingguy.php` role, but would take absolutely forever on the Pi).
  - Pin php* packages to the Debian `jessie` repo sources (via [pinning](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=47944#p375741)).
  - See if the [dotdeb](https://extremeshok.com/5946/debian-7-php-5-5-upgrade-from-php-5-4/) or [ondrej](https://launchpad.net/~ondrej/+archive/ubuntu/php5) PPAs work with the Raspberry Pi (doubtful, as I'm guessing they're only built for `x64` architectures, but worth a try.

I need to create a sister issue in the `drupal-pi` project, since this 5.5+ will soon become a hard requirement for Drupal 8...

geerlingguy, 2015-07-05:
After playing around with all the above approaches, it looks like compiling from source might be the easiest way to go, even though it takes a bit of time to compile PHP on the Pi :(

geerlingguy, 2015-07-05:
On second thought, after having waited about 45 minutes to try building PHP from source, it might be easiest to use `jessie` and pin to that repo for `php*` packages. Trying that now... Can't easily get PHP 5.6 or 7.x that way, but at least we can get to 5.5 without waiting an hour or two :P

geerlingguy, 2015-07-06:
Nope, still a no-go, unless I want to upgrade everything to Jessie, which I don't really at this time. I'd rather stick to a more stable set of packages from Raspbian mainline... hopefully they upgrade to Jessie/8 sometime soon, but I won't hold my breath.

Back to building from source—I'm timing the process this time.

FelicianoTech, 2015-07-06:
What about cross-compiling from an x86 machine?

https://www.raspberrypi.org/documentation/linux/kernel/building.md
http://openframeworks.cc/setup/raspberrypi/Raspberry-Pi-Cross-compiling-guide.html

geerlingguy, 2015-07-06:
It took about 45 minutes just to do the clone, configure, compile and install... but:

```
$ php -v
PHP 5.5.26 (cli) (built: Jul  6 2015 02:38:11) 
Copyright (c) 1997-2015 The PHP Group
Zend Engine v2.5.0, Copyright (c) 1998-2015 Zend Technologies
```

Next up, I'm going to wipe it all, and overnight let it compile PHP HEAD and see what happens :)

Configuration changes incoming... and there's no way you'll want to attempt compiling on a B+.

geerlingguy, 2015-07-06:
@FelicianoTech - That's a definite long-term possibility. The compile time is pretty annoyingly long. However, I wouldn't want people to have to set up a separate computer/VM to do the cross-compile just to save a few minutes—my goal is to have a configuration that runs D8 optimally on a Raspberry Pi with as little fuss as possible.

At this point, I'm getting everything to install flawlessly with PHP 5.4.x, but I'm testing 5.5, 5.6 and 7.0 to see if they're all stable as compiled on the Pi. And unfortunately, there are no package repositories that I can find with ARM packages for PHP > 5.4 on Debian Wheezy, and the only other option is to do a full dist upgrade to Jessie, which I'd rather not do at this time.

I'm guessing in 6 months to a year, Raspbian's default image/repos will have 5.5 in them, so it's just a waiting game. And since this project is more for the fun of it (there's no way I'd deliver production Drupal traffic to a Pi 2 ;-), I'm okay with compilation taking quite a while, as long as it's flexible (which encourages experimentation) and easy enough to use the playbook.

geerlingguy, 2015-07-06:
I've punted around the idea of using my Dramble stack to build/manage my own RPi-specific apt repo for later versions of PHP and maybe a couple other tools... but that's just a bit beyond what I'd be willing to maintain (on top of the bajillion other OSS stuff I'm involved with :P ).

geerlingguy, 2015-07-06:
I've added some actual code for the `drupal-pi` project in https://github.com/geerlingguy/drupal-pi/pull/4 — I'm still running into little issues here and there (especially with 5.6.10 and 7.0.0alpha2), so I'm trying to get things more stable before promoting that to master, and then merging it into this upstream project (especially since compilation takes 30-45 minutes _minimum_—no way I want someone to have to sit and wait on 4 Pis to see that an unstable release/process requires reimaging everything!).

geerlingguy, 2015-07-27:
Issues are no longer issues, so I'm going to work on getting up to 5.6.11 stable for now.

commit f8c9fefd (2015-07-27): Fixes #51: Use PHP 5.6 (5.4 is only available version in Wheezy).
closed 2015-07-27 by geerlingguy