geerlingguy/raspberry-pi-dramble #58: Disable HDMI port on boot (save ~30 mA per Pi)

geerlingguy, 2015-11-29:
See: http://www.midwesternmac.com/blogs/jeff-geerling/raspberry-pi-zero-power

Eightyplus, 2016-10-26:
Would it be possible to make this optional? Say I want to enable HDMI if 'd' was pressed during boot?

geerlingguy, 2016-10-26:
@Eightyplus - I would make it configurable in the Dramble Ansible configuration, of course, but it would require more work to make it togglable at boot time. You'd have to have some sort of shell script that would run under certain conditions and not under others to do that...

Eightyplus, 2016-10-27:
[Wonder why there is not a generic way of forwarding keyboard tabs during boot...]
My router died and I had it replaced, that and in some other cases it would be nice with a optional disable. 
To avoid to much work. I could check if the wifi-dongle is plugged in, if not then boot with HDMI.

Something like:
```shell
if [ ! -z 'dmesg | grep rt2800usb']; then
     /usr/bin/tvservice -o
fi
```

(the ' should be replaced by `)

Or `grep ieee80211` or `grep wlan0` or `modprobe`? Will have to test when I get home.

Eightyplus, 2016-10-28:
Nice, i can check on the wifi dongle (`ieee80211` or `wlan0`) with `dmesg`! 
_Note: checking for rt2800usb does not work since it always loads._

```bash
wifi_loaded=`dmesg | grep "ieee80211"`

if [ -n "$wifi_loaded" ]; then
     /usr/bin/tvservice -o
fi

```

rin67630, 2018-05-12:
Would it be possible to check if a keyboard is present instead?

Eightyplus, 2018-05-12:
Yeah I guess so, I would start of by logging dmesg with and without keyboard. Add the following to 
```
dmesg >  `date +%Y-%m-%d-%H-%M`.log
```

Your log should show something like `# [  123.141469] usb 2-2: Product: USB Keykoard`, but I guess that is you to verify/find. Then your command would be something like:

```
keyboard_present=`dmesg | grep -i "usb keyboard"`

if [ -n "$keyboard_present" ]; then
     /usr/bin/tvservice -o
fi
```