geerlingguy/raspberry-pi-dramble #65: Fix rpcbind not enabled at startup with Raspbian Jessie Lite

geerlingguy, 2016-03-14:
Apparently the version of rpcbind that's currently shipping with Raspbian Jessie Lite is causing some issues at startup that cause the NFS server running on the first webserver to fail.

If I run `sudo systemctl status nfs-kernel-server` it shows the following error:

```
Not starting: portmapper is not running ...
```

I tried working around it by manually adding in the following in the `nfs.yml` playbook, but that didn't work after the first provision (e.g. it would start failing after a reboot, and the NFS shares would not be shared, so mounts fail):

```yaml
- name: Ensure rpcbind is running for Raspbian.
  service: name=rpcbind state=started enabled=yes
  register: rpcbind_start
  when: deploy_target == 'pi'

- name: Force restart NFS if rpcbind wasn't started.
  service: "name={{ nfs_server_daemon }} state=restarted"
  when: rpcbind_start.changed
```

So the next step is to see about installing the latest release of rpcbind, which might work correctly with systemd/Jessie... See: https://www.raspberrypi.org/forums/viewtopic.php?t=128816&p=862161#p862161

geerlingguy, 2016-03-16:
Looks like the rpcbind install that is loaded by default in Raspbian _might_ be at fault, so I'm going to test purging it before starting the NFS install (as part of the earliest config for the webservers), and we'll see if that works... See: http://raspberrypi.stackexchange.com/a/24914/6506

closed 2016-03-16 by geerlingguy

daurigi, 2017-01-15:
This solution does not survive a reboot on Linux diluvo 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux.

geerlingguy, 2017-01-16:
@daurigi - D'oh! Can you open a new issue for that?

daurigi, 2017-01-16:
The problem is with the runlevels for nfs-common and rpcbind. These are the most clear (and working solutions) I found. http://lesca.me/archives/debian-or-raspbian-nfs-kernel-server-not-starting-portmapper-is-not-running.html and, http://raspberrypi.stackexchange.com/a/54472. As David mentions in his comment, a NFS upgrade might get your problem back.

Scalextrix, 2017-02-15:
Subscribed, seeing same issue on Pi2 and Pi3 with Jesie-lite

artmg, 2017-02-28:
Hi, I had similar problems with Jesse lite 2016-02-09 and 2017-02-16 (latest uses rpcbind 0.2.1-6). The server-side fix I used is based on something that hasn't yet made it downstream to production, but see the references below for info.
```
# credit [https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=149002&p=980161]
# Original [https://lists.debian.org/debian-devel/2014/05/msg00618.html]

cat <<EOF | sudo tee -a /etc/systemd/system/nfs-common.services
[Unit]
Description=NFS Common daemons
Wants=remote-fs-pre.target
DefaultDependencies=no

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/etc/init.d/nfs-common start
ExecStop=/etc/init.d/nfs-common stop

[Install]
WantedBy=sysinit.target
EOF

cat <<EOF | sudo tee -a /etc/systemd/system/rpcbind.service
[Unit]
Description=RPC bind portmap service
After=systemd-tmpfiles-setup.service
Wants=remote-fs-pre.target
Before=remote-fs-pre.target
DefaultDependencies=no

[Service]
ExecStart=/sbin/rpcbind -f -w
KillMode=process
Restart=on-failure

[Install]
WantedBy=sysinit.target
Alias=portmap
EOF

sudo systemctl enable nfs-common
sudo systemctl enable rpcbind
sudo reboot 
```
Hope this helps :)

commit 69d387bd (2016-03-16): Fixes #65: Fix rpcbind not enabled at startup with Raspbian Jessie Lite.