geerlingguy/raspberry-pi-dramble #82: Provisioning

thall005, 2016-07-31:
I am Currently running Linux Mint 18 Sarah  to provision my dramble. When I enter the code i get the following Message
 $ ansible-galaxy -r playbooks/requirements.yml --force  
Usage: ansible-galaxy [delete|import|info|init|install|list|login|remove|search|setup] [--help] [options] ...

Options:
  -h, --help     show this help message and exit
  -v, --verbose  verbose mode (-vvv for more, -vvvv to enable connection
                 debugging)
  --version      show program's vers
ERROR! Missing required action

WassimAttar, 2016-10-14:
The command line written there is incorrect
http://www.pidramble.com/wiki/setup/provision
ansible-galaxy -r playbooks/requirements.yml --force

The correct one is
ansible-galaxy install -r playbooks/requirements.yml --force

@geerlingguy it will be great if you correct it.

geerlingguy, 2016-10-17:
@WassimAttar - Oh, sorry about that! You're quite correct, and I've updated the Wiki page. It may take an hour or two to show (since the site is cached via CloudFlare), but thanks so much for the feedback.

closed 2016-10-17 by geerlingguy