geerlingguy/raspberry-pi-dramble #86: microsd-benchmark.sh without sudo

gddeen, 2017-03-15:
I hope this is right repos...

the microsd-benchmark.sh script, if invoked without sudo (That's kinda ok... just installs and iotest fail, BUT SUDO_USER is blank.

That causes getent to return every root directory for ever system ID and then go erase all of them and the last one with '/test' on the end... Kinda of an unlucky today, I am...

s-e-a-n-s, 2017-04-20:
(landed here chasing the same issue)  This script is referenced several places on the web, so I'd like to help make it a little more forgiving.  Running the script directly as root (vs. through sudo) can wipe your SD card.

@geerlingguy -   I'm going to submit a pull request with a simple if [ -z "$SUDO_USER" ]; then ....; exit 1; - this will prevent accidental 'rm -f /root ...' calls.

If this script is being more widely used, I'm happy to add a bit more, such as..
 - moving dd of to /tmp  (using mktemp) (easy/safe sidestep to allow sudo and non-sudo access)
 - if present, use iozone already present in path,
 - (maybe) option command line param for iozone size.

Please let me know if you have any questions/concerns/feedback.

geerlingguy, 2017-04-20:
Yikes, yeah @s-e-a-n-s it sounds like that would be a very good idea.

commit 75c51729 (2017-04-21): Merge pull request #91 from s-e-a-n-s/benchmark-mods

Fixes #86: require SUDO_USER to prevent downstream issues
commit c7419034 (2017-04-21): Fixes #86: require SUDO_USER to prevent downstream issues
closed 2017-04-21 by geerlingguy