geerlingguy/raspberry-pi-dramble #88: Copy munin configurations fails

aaronkjones, 2017-03-19:
I changed the inventory file to the following:

> [balancer]
> 192.168.1.50
> 
> [webservers]
> 192.168.1.51
> 192.168.1.52
> 
> [database]
> 192.168.1.5
> 
> [dramble:children]
> balancer
> webservers
> database
> 
> [dramble:vars]
> ansible_ssh_user=pi
> deploy_target=pi

The error I get when I run the "ansible-playbook -i inventory main.yml" command:
```
TASK [geerlingguy.munin : Copy munin configurations.] **************************
ok: [192.168.1.50] => (item={u'dest': u'/etc/munin/munin.conf', u'src': u'munin.conf.j2'})
failed: [192.168.1.50] (item={u'dest': u'/etc/munin/munin-conf.d/hosts.conf', u'src': u'hosts.conf.j2'}) => {"failed": true, "item": {"dest": "/etc/munin/munin-conf.d/hosts.conf", "src": "hosts.conf.j2"}, "msg": "AnsibleUndefinedVariable: [{u'extra': [], u'name': u'dramble;bal1.{{ site_url }}', u'address': u'127.0.0.1'}, {u'extra': [], u'name': u'dramble;www1.{{ site_url }}', u'address': u'{{ groups.webservers[0] }}'}, {u'extra': [], u'name': u'dramble;www2.{{ site_url }}', u'address': u'{{ groups.webservers[1] }}'}, {u'extra': [], u'name': u'dramble;db1.{{ site_url }}', u'address': u'{{ groups.database[0] }}'}, {u'extra': [], u'name': u'dramble;db2.{{ site_url }}', u'address': u'{{ groups.database[1] }}'}]: list object has no element 1"}
        to retry, use: --limit @/home/akj/Downloads/raspberry-pi-dramble/main.retry
```

aaronkjones, 2017-03-19:
This is the section that was giving me trouble,

> # Munin configuration.

> munin_admin_user: admin
> munin_admin_password: secret
> munin_hosts:
>   - { name: "dramble;bal1.{{ site_url }}", address: "127.0.0.1", extra: [] }
>   - { name: "dramble;www1.{{ site_url }}", address: "{{ groups.webservers[0] }}", extra: [] }
>   - { name: "dramble;www2.{{ site_url }}", address: "{{ groups.webservers[1] }}", extra: [] }
>   - { name: "dramble;db1.{{ site_url }}", address: "{{ groups.database[0] }}", extra: [] }
>   - { name: "dramble;db2.{{ site_url }}", address: "{{ groups.database[1] }}", extra: [] }

aaronkjones, 2017-03-19:
I also had to modify /playbooks/database/vars.yml because I do not have enough raspberry pis to make a secondary mirrored db

geerlingguy, 2017-04-21:
At this time the playbook isn't flexible enough to handle a different number of Pis than what I originally based it on (5 Pis), so it's understandable that you'd need to make some modifications to use just one DB server or something like just one webserver.

Also, note that the drupal-pi project (linked in the README) could also be used if you wanted to use just one or two Pis total.

coto2448, 2017-04-22:
aaronkjones@ -  I got the same set up, made it work besides the issue noted back on "issues". But believe i passed this part with no problem, just delete that;
{ name: "dramble;db2.{{ site_url }}", address: "{{ groups.database[1] }}", extra: [] }

and anything to do with the 2DB, same for LEDs since i dont have the equipment but will retrace my steps once i get the parts.

geerlingguy@ - thanks for the tip will take a look at the README file too.