geerlingguy/raspberry-pi-dramble #90: FAILED! => {"changed": false, "failed": true, "msg": "Failed to find required executable git"}

coto2448, 2017-04-10:
having issues here;

TASK [Check out Drupal to the docroot.] ****************************************
fatal: [10.0.0.62]: FAILED! => {"changed": false, "failed": true, "msg": "Failed to find required executable git"}
fatal: [10.0.0.61]: FAILED! => {"changed": false, "failed": true, "msg": "Failed to find required executable git"}

Notes; i dont have leds so i have mod/del or ignored those files regarding "rgb" also have only 4 pis; LB, 2 webs and 1 DB.

coto2448, 2017-04-10:
Managed to get around that by;

# Uncomment these lines to install Drupal 8 from drupal.org instead of the demo
# Drupal 8 project (override vars from `playbooks/drupal/vars/main.yml`).
# drupal_repo: http://git.drupal.org/project/drupal.git
# drupal_deploy_strategy: "tarball"
# drupal_version: "8.2.6"

but now stuck on this task;

- name: Run composer install if composer.json is present.
  command: >
    composer install
    chdir={{ drupal_docroot }}
  when: drupal_core_composer_file.stat.exists == true
  become: yes
  become_user: www-data

the error i get in verbose;

fatal: [10.0.0.61]: FAILED! => {
    "changed": false, 
    "cmd": "composer install", 
    "failed": true, 
    "invocation": {
        "module_args": {
            "_raw_params": "composer install", 
            "_uses_shell": false, 
            "chdir": "/var/www/drupal", 
            "creates": null, 
            "executable": null, 
            "removes": null, 
            "warn": true
        }, 
        "module_name": "command"
    }, 
    "msg": "[Errno 2] No such file or directory", 
    "rc": 2
}

geerlingguy, 2017-04-21:
@coto2448 - Hmm, I wonder if composer isn't installed, or if `/var/www/drupal` doesn't exist. It seems like it would be one of those two things.

coto2448, 2017-04-21:
well its not in my localhost but it is in my web pis;
sending an adhoc; "find  /var/www/drupal -print -quit":
10.0.0.61 | SUCCESS | rc=0 >>
/var/www/drupal

10.0.0.62 | SUCCESS | rc=0 >>
/var/www/drupal

geerlingguy, 2017-04-22:
In that case, can you run an adhoc with `which composer`? Maybe it's not seeing composer...

coto2448, 2017-04-22:
So "which composer" is not pulling anything, find either, or composer --version.

Here is some output with "locate composer" (pretty long so brace yourself, for both web servers I attached it instead):

[composer_output.txt](https://github.com/geerlingguy/raspberry-pi-dramble/files/949375/composer_output.txt)

Thanks for taking the time to help.

geerlingguy, 2017-04-23:
Interesting... I just noticed I have the composer role in requirements here: https://github.com/geerlingguy/raspberry-pi-dramble/blob/4d6826a9da546b5a35ccaec336369d0a354b4709/playbooks/requirements.yml#L7

But I'm not actually calling it in my Drupal playbook!

Can you add to this file (https://github.com/geerlingguy/raspberry-pi-dramble/blob/master/playbooks/drupal/main.yml) the code block below, between the `pre_tasks` and `tasks` section, and see if that fixes it for you?

```yaml
  roles:
    - geerlingguy.composer
```

coto2448, 2017-04-24:
@geerlingguy - So i think it may have helped, or now this may be part of the problem?: 

TASK [geerlingguy.composer : Download Composer installer.] *********************
ok: [10.0.0.61]
ok: [10.0.0.62]

TASK [geerlingguy.composer : Run Composer installer.] **************************
fatal: [10.0.0.61]: FAILED! => {"changed": false, "cmd": "php composer-installer.php", "failed": true, "msg": "[Errno 2] No such file or directory", "rc": 2}
fatal: [10.0.0.62]: FAILED! => {"changed": false, "cmd": "php composer-installer.php", "failed": true, "msg": "[Errno 2] No such file or directory", "rc": 2}
	to retry, use: --limit @/etc/ansible/playbooks/drupal/main.retry

So when i retry, in verbose;

fatal: [10.0.0.61]: FAILED! => {
    "changed": false, 
    "cmd": "php composer-installer.php", 
    "failed": true, 
    "invocation": {
        "module_args": {
            "_raw_params": "php composer-installer.php ", 
            "_uses_shell": false, 
            "chdir": "/tmp", 
            "creates": null, 
            "executable": null, 
            "removes": null, 
            "warn": true
        }
    }, 
    "msg": "[Errno 2] No such file or directory", 
    "rc": 2
}
fatal: [10.0.0.62]: FAILED! => {
    "changed": false, 
    "cmd": "php composer-installer.php", 
    "failed": true, 
    "invocation": {
        "module_args": {
            "_raw_params": "php composer-installer.php ", 
            "_uses_shell": false, 
            "chdir": "/tmp", 
            "creates": null, 
            "executable": null, 
            "removes": null, 
            "warn": true
        }
    }, 
    "msg": "[Errno 2] No such file or directory", 
    "rc": 2
}

just wanted to check in one of my Pis;

pi3-0:/tmp $ ls
composer-installer.php

it is there, or should it say "php composer-installer.php "  ?

Here i added my version of /drupal/main.yml  below "attached"( I changed and deleted a few things since i dont have the LEDs and  indented handlers  so that it can read when using  $ ansible-playbook -i inventory playbooks/drupal/main.yml).

[main.yml.txt](https://github.com/geerlingguy/raspberry-pi-dramble/files/952807/main.yml.txt)