geerlingguy/raspberry-pi-dramble #93: The ansible-role-security is failing.

Unifex, 2017-06-29:
Hello,

I'm _very_ new to ansible so I'm not sure how to debug this.  From what I can tell though the firewall is locking down the Pi to the point where it can no longer reach the apt repo to download and install fail2ban.  I'm not sure if this is an issue with raspberry-pi-dramble, ansible-role-firewall or ansible-role-security.

This is the output though.  I've rebuilt the initial Pi (bal1) twice now and have had the same result.

If there is anything more that I can provide to assist debugging this please let me know.

Cheers,
Gold

```
unifex@shipnet:~/Projects/raspberry-pi-dramble$ ansible-playbook -i inventory main.yml
[DEPRECATION WARNING]: Specifying include variables at the top-level of the task is deprecated. Please see:

http://docs.ansible.com/ansible/playbooks_roles.html#task-include-files-and-encouraging-reuse

 for currently supported syntax regarding 
included files and variables. This feature will be removed in a future release. Deprecation warnings can be disabled by setting 
deprecation_warnings=False in ansible.cfg.

PLAY ***************************************************************************

TASK [setup] *******************************************************************
ok: [192.168.0.160]

TASK [Ensure sudo is installed.] ***********************************************
ok: [192.168.0.160]

TASK [geerlingguy.firewall : Ensure iptables is installed.] ********************
ok: [192.168.0.160]

TASK [geerlingguy.firewall : Flush iptables the first time playbook runs.] *****
changed: [192.168.0.160]

TASK [geerlingguy.firewall : Copy firewall script into place.] *****************
changed: [192.168.0.160]

TASK [geerlingguy.firewall : Copy firewall init script into place.] ************
skipping: [192.168.0.160]

TASK [geerlingguy.firewall : Copy firewall systemd unit file into place (for systemd systems).] ***
changed: [192.168.0.160]

TASK [geerlingguy.firewall : Ensure the firewall is enabled and will start on boot.] ***
changed: [192.168.0.160]

TASK [geerlingguy.firewall : include] ******************************************
skipping: [192.168.0.160]

TASK [geerlingguy.security : Include OS-specific variables.] *******************
ok: [192.168.0.160]

TASK [geerlingguy.security : include] ******************************************
skipping: [192.168.0.160]

TASK [geerlingguy.security : include] ******************************************
included: /etc/ansible/roles/geerlingguy.security/tasks/fail2ban-Debian.yml for 192.168.0.160

TASK [geerlingguy.security : Install fail2ban.] ********************************
fatal: [192.168.0.160]: FAILED! => {"cache_update_time": 0, "cache_updated": false, "changed": false, "failed": true, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"   install 'fail2ban'' failed: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/f/fail2ban/fail2ban_0.8.13-1_all.deb  Could not resolve 'mirrordirector.raspbian.org'\n\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/pyinotify/python-pyinotify_0.9.4-1_all.deb  Could not resolve 'mirrordirector.raspbian.org'\n\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "stderr": "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/f/fail2ban/fail2ban_0.8.13-1_all.deb  Could not resolve 'mirrordirector.raspbian.org'\n\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/pyinotify/python-pyinotify_0.9.4-1_all.deb  Could not resolve 'mirrordirector.raspbian.org'\n\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following extra packages will be installed:\n  python-pyinotify\nSuggested packages:\n  python-gamin mailx python-pyinotify-doc\nThe following NEW packages will be installed:\n  fail2ban python-pyinotify\n0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.\nNeed to get 192 kB of archives.\nAfter this operation, 713 kB of additional disk space will be used.\nErr http://mirrordirector.raspbian.org/raspbian/ jessie/main fail2ban all 0.8.13-1\n  Could not resolve 'mirrordirector.raspbian.org'\nErr http://mirrordirector.raspbian.org/raspbian/ jessie/main python-pyinotify all 0.9.4-1\n  Could not resolve 'mirrordirector.raspbian.org'\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following extra packages will be installed:", "  python-pyinotify", "Suggested packages:", "  python-gamin mailx python-pyinotify-doc", "The following NEW packages will be installed:", "  fail2ban python-pyinotify", "0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.", "Need to get 192 kB of archives.", "After this operation, 713 kB of additional disk space will be used.", "Err http://mirrordirector.raspbian.org/raspbian/ jessie/main fail2ban all 0.8.13-1", "  Could not resolve 'mirrordirector.raspbian.org'", "Err http://mirrordirector.raspbian.org/raspbian/ jessie/main python-pyinotify all 0.9.4-1", "  Could not resolve 'mirrordirector.raspbian.org'"]}

RUNNING HANDLER [geerlingguy.firewall : restart firewall] **********************

PLAY RECAP *********************************************************************
192.168.0.160              : ok=9    changed=4    unreachable=0    failed=1
```

geerlingguy, 2017-06-30:
That could also be a DNS issue; can you log into the Pi and do a `ping www.google.com` or even a `dig` or something to see if DNS works?

n3tsec, 2017-07-01:
I had the same issue few minutes ago. In my case I had OpenDNS servers instead of google when I changed to the 8.8.4.4 8.8.8.8 and rebooted the pi's it worked. 

ssh your bal1 pi and make sure you can # **sudo apt-get update** or ping google.com 

I hope this help.

Unifex, 2017-07-06:
Hello Jeff,

I can confirm that the Pis were all able to reach the net prior to starting the ansible call.  During network setup I was able to apt update/upgrade them all successfully.

This also happened on the first Pi and the process bailed on that one.  From that Pi I'm unable to reach out to the net at all. I'll check when I get home and see if I can reach things on the local network.

I'll also double check @n3tsec's suggestion.

geerlingguy, 2017-07-06:
Hmm, maybe one of the networking changes knocks out DNS :/

I didn't have issues the last time I set things up on the Pis themselves, but it's been a couple months. Raspbian seems to change things up on a monthly basis!

Unifex, 2017-07-06:
One of the things that I did when setting these up was to apt update/upgrade.  I'll re-flash the card and try again this weekend.

Regarding the earlier comment, I'm checked options for getting out of the Pi and tried to ping a number of domains and IP addresses directly.  No luck.

It is still feeling like it's an IPTables thing.  These are the current rules:
`pi@bal1:~ $ sudo iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere            
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:ssh
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:http
ACCEPT     icmp --  anywhere             anywhere            
ACCEPT     udp  --  anywhere             anywhere             udp spt:ntp
ACCEPT     all  --  anywhere             anywhere             state RELATED,ESTABLISHED
LOG        all  --  anywhere             anywhere             limit: avg 15/min burst 5 LOG level debug prefix "Dropped by firewall: "
DROP       all  --  anywhere             anywhere            

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     udp  --  anywhere             anywhere             udp dpt:ntp`

From my very limited understanding of iptables that Chain OUTPUT section looks very limiting.

Hopefully this helps.

geerlingguy, 2018-05-30:
For OUTPUT, the default policy is ACCEPT, so no matter what, if no other rule denies, any traffic is accepted.

Closing this out for now, but please let me know if there are still any other issues.

closed 2018-05-30 by geerlingguy