geerlingguy/raspberry-pi-dramble #94: Working on the project. NEED HELP WITH IT! THANK!

n3tsec, 2017-07-01:
Hello Geerlingguy,

First of all, thanks a lot for this project it's awesome!

I have been trying to do this for a week now with not success, however I keep trying by looking on forums for the issues that I encounter before posting in here, so i can fixed it myself. There is been a point that I have re-image the SD's 2 times so far O_o.   

Anyways, If you could help me through my installation I will really appreciate. I will be posting the issues / errors that I encounter during my installation until I done.

I'm using **4 pi's**  just to letting you know my setup bal1 / www1 / www2 / db1. 

First error:
TASK [geerlingguy.php : Clone the PHP repository.] *****************************
fatal: [192.168.1.143]: FAILED! => {"changed": false, "cmd": "/usr/bin/git checkout --force php-7.0.17", "failed": true, "msg": "Failed to checkout php-7.0.17", "rc": 1, "stderr": "error: pathspec 'php-7.0.17' did not match any file(s) known to git.\n", "stdout": "", "stdout_lines": []}
fatal: [192.168.1.142]: FAILED! => {"changed": false, "cmd": "/usr/bin/git checkout --force php-7.0.17", "failed": true, "msg": "Failed to checkout php-7.0.17", "rc": 1, "stderr": "error: pathspec 'php-7.0.17' did not match any file(s) known to git.\n", "stdout": "", "stdout_lines": []}

Please let me know what should I do here.

Regards,

n3tsec, 2017-07-01:
Fixed by changing _depth: 1_  I followed this post here: https://github.com/ansible/ansible-modules-core/issues/5457

geerlingguy, 2017-07-01:
I can't promise to be too timely, especially this weekend, but I'll at least work on a little testing for my own satisfaction to see if I can run a build locally (with just Vagrant).

n3tsec, 2017-07-01:
I believe that I'm almost done. I just encounter this error that idk what to do....

TASK [Install Drupal.] *********************************************************
skipping: [192.168.1.143]
fatal: [192.168.1.142]: FAILED! => {"changed": true, "cmd": ["drush", "si", "standard", "-y", "--site-name=Site", "--account-name=drupal", "--account-pass=Password"], "delta": "0:00:00.875055", "end": "2017-07-01 04:39:03.679309", "failed": true, "rc": 1, "start": "2017-07-01 04:39:02.804254", "stderr": "PHP Warning:  PHP Startup: Unable to load dynamic library '/opt/php/lib/php/extensions/no-debug-non-zts-20151012/apcu.so' - /opt/php/lib/php/extensions/no-debug-non-zts-20151012/apcu.so: cannot open shared object file: No such file or directory in Unknown on line 0\nCommand site-install needs a higher bootstrap level to run - you will\u001b[31;40m\u001b[1m[error]\u001b[0m\nneed to invoke drush from a more functional Drupal environment to run\nthis command.\nThe drush command 'si standard' could not be executed.               \u001b[31;40m\u001b[1m[error]\u001b[0m", "stdout": "", "stdout_lines": [], "warnings": []}

Can you help? Thanks!

geerlingguy, 2017-07-01:
The error seems to be:

```
PHP Startup: Unable to load dynamic library '/opt/php/lib/php/extensions/no-debug-non-zts-20151012/apcu.so'
```

That means we're probably installing apcu incorrectly somewhere along the way :/

n3tsec, 2017-07-01:
Yea. I can't find apcu.so anywhere on the pi's....  The only 2 files inside there are opcache.a  opcache.so what should I do... :/