geerlingguy/raspberry-pi-dramble #97: The error was: 'dict object' has no attribute u'b8: 27:

webmanz, 2017-09-13:
Hi Jeff,

Thank's for the "Ansible for..” book & your amazing open source pidramble project.

I'm having trouble running Ansible 2.3.2.0 on my MacOS Sierra PB:
ansible-playbook -i inventory main.yml
…..
set_fact:
        dramble_hostname: "{{ mac_address_mapping[dramble_mac_address].name }}”
...
"FAILED! => {" Failed ": true," msg ":" the field 'args' has an invalid value, which appears to include a variable that is undefined. The error was: 'dict object' has no attribute u'b8: 27: eb: ec: ff: 3d '\ n "
"{{mac_address_mapping [dramble_mac_address] .name}}"
What I ...understand Ansible can not read the ”mac_address_mapping” syntax, has the syntax changed?
Or Am I missing some additional Ansible extension  …?

I’ve followed your instructions…  carefully?smile.png

Best regards / p

geerlingguy, 2017-09-19:
It looks like that might be a quoting issue—can you paste in an example of how you changed the networking settings? It should be like:

```yaml
mac_address_mapping:
  "d8:eb:97:b3:88:27":
    name: bal1.pidramble.com
    ip: "10.0.1.60"
  ...
```

Notice the quotes around the entire d8:eb:97... part — without those quotes, the YAML parser will get very confused by all the colons.

webmanz, 2017-09-19:
This is pat of the changes:
mac_address_mapping:
  "B8:27:EB:EF:7D:B1":
    name: bal1.pidramble.com
    ip: "10.0.1.60"
  "bB8:27:EB:EC:FF:3D":
    name: www1.pidramble.com
    ip: "10.0.1.61"
....
/p

webmanz, 2017-09-20:
Resolved! "bB8:27:EB:EC:FF:3D": to many bees:) and now all in lowercase letters, do not know if it matters /p

geerlingguy, 2018-01-23:
Perfect! Glad you could figure it out. And watch out for the 🐝 s!

closed 2018-01-23 by geerlingguy