geerlingguy/raspberry-pi-dramble #99: Issues found during Provision the Raspberry Pis Steps

click2tman, 2017-12-08:
It seems things are failing during the PHP install with the following.
TASK [geerlingguy.php : Ensure dependencies for building from source are installed (Debian).]
failed: [192.168.0.202] (item=[u'build-essential', u'autoconf', u'automake', u'libtool', u'bison', u'pkg-config', u're2c', u'libxml2-dev', u'libcurl4-openssl-dev', u'libbz2-dev', u'libjpeg-dev', u'libpng12-dev', u'libxpm-dev', u'libfreetype6-dev', u'libgmp3-dev', u'libmcrypt-dev', u'libmysqlclient-dev', u'libpspell-dev', u'librecode-dev', u'libssl-dev']) => {"changed": false, "failed": true, "item": ["build-essential", "autoconf", "automake", "libtool", "bison", "pkg-config", "re2c", "libxml2-dev", "libcurl4-openssl-dev", "libbz2-dev", "libjpeg-dev", "libpng12-dev", "libxpm-dev", "libfreetype6-dev", "libgmp3-dev", "libmcrypt-dev", "libmysqlclient-dev", "libpspell-dev", "librecode-dev", "libssl-dev"], "msg": "No package matching 'libmysqlclient-dev' is available"}
failed: [192.168.0.203] (item=[u'build-essential', u'autoconf', u'automake', u'libtool', u'bison', u'pkg-config', u're2c', u'libxml2-dev', u'libcurl4-openssl-dev', u'libbz2-dev', u'libjpeg-dev', u'libpng12-dev', u'libxpm-dev', u'libfreetype6-dev', u'libgmp3-dev', u'libmcrypt-dev', u'libmysqlclient-dev', u'libpspell-dev', u'librecode-dev', u'libssl-dev']) => {"changed": false, "failed": true, "item": ["build-essential", "autoconf", "automake", "libtool", "bison", "pkg-config", "re2c", "libxml2-dev", "libcurl4-openssl-dev", "libbz2-dev", "libjpeg-dev", "libpng12-dev", "libxpm-dev", "libfreetype6-dev", "libgmp3-dev", "libmcrypt-dev", "libmysqlclient-dev", "libpspell-dev", "librecode-dev", "libssl-dev"], "msg": "No package matching 'libmysqlclient-dev' is available"}

geerlingguy, 2018-01-23:
Interesting... it should be there, but maybe sources have changed recently. I'll have to boot a test stack to see what's going on.

brianj87, 2018-05-16:
libmysqlclient-dev has been changed to default-libmysqlclient-dev on stretch. Not an issue on jessie.
You will also run into an error with newer openssl packages on stretch if you manage to get past this error.

geerlingguy, 2018-05-30:
@brianj87 - Thanks for the updates! I may try to get the packages updated soon...