geerlingguy/raspberry-pi-dramble #108: Deploy a local Docker registry inside Kubernetes cluster

geerlingguy, 2018-06-04:
See: [Deploying a Docker Registry, with persistent storage and local-only access, on your Kubernetes cluster](https://medium.com/@jmarhee/deploying-a-docker-registry-with-persistent-storage-and-local-only-access-on-your-kubernetes-1c6470d3611c).

This will allow #104 to build and store the Drupal image(s) locally inside the cluster, so we don't need to either do the build/run/hosting locally on a host computer, or on a separate node. It's also a pattern I think would help a lot with many microservices-oriented architectures which can have a private self-contained registry running inside the same cluster.

geerlingguy, 2018-06-17:
See specifically:

  - Docker registry creation docs on how to use a self-signed cert: [Get a Certificate](https://docs.docker.com/registry/deploying/#get-a-certificate).
  - [kubectl create secret docker-registry](https://kubernetes-v1-4.github.io/docs/user-guide/kubectl/kubectl_create_secret_docker-registry/)
  - [Running a Secure Registry on Kubernetes](https://container-solutions.com/running-secure-registry-kubernetes/)

geerlingguy, 2018-06-25:
So, got most of this working currently. However, I need to work on having the docker registry run with ingress or something, so any node in the cluster can pull from / push to it. Or figure out a different docker-in-docker solution for building and pushing images...

But probably better to build a full-fledged/good docker registry solution that could even be exposed outside the cluster, with proper ingress.

geerlingguy, 2018-06-26:
To test the registry:

```
docker pull alpine:latest
docker tag alpine:latest registry.pidramble.test/alpine:latest
docker push registry.pidramble.test/alpine:latest
```

geerlingguy, 2018-06-26:
Hmm... from my Mac, it looks like requests aren't getting through correctly:

```
 22:43:43 ~ $ curl -vvvv --insecure https://registry.pidramble.test/debug/health
*   Trying 192.168.77.4...
* TCP_NODELAY set
* Connection failed
* connect to 192.168.77.4 port 443 failed: Connection refused
* Failed to connect to registry.pidramble.test port 443: Connection refused
* Closing connection 0
curl: (7) Failed to connect to registry.pidramble.test port 443: Connection refused
 22:44:03 ~ $ curl -vvvv http://registry.pidramble.test/debug/health
*   Trying 192.168.77.4...
* TCP_NODELAY set
* Connected to registry.pidramble.test (192.168.77.4) port 80 (#0)
> GET /debug/health HTTP/1.1
> Host: registry.pidramble.test
> User-Agent: curl/7.54.0
> Accept: */*
> 
< HTTP/1.1 503 Service Unavailable
< Date: Tue, 26 Jun 2018 03:47:15 GMT
< Content-Length: 19
< Content-Type: text/plain; charset=utf-8
< 
* Connection #0 to host registry.pidramble.test left intact
```

geerlingguy, 2018-06-26:
Also from inside...

```
root@kube1:~# curl -vvvv https://10.103.37.154/debug/health
*   Trying 10.103.37.154...
* TCP_NODELAY set
* Connected to 10.103.37.154 (10.103.37.154) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=docker-registry
*  start date: Jun 25 17:23:41 2018 GMT
*  expire date: Jun 22 17:23:41 2028 GMT
*  subjectAltName does not match 10.103.37.154
* SSL: no alternative certificate subject name matches target host name '10.103.37.154'
* Curl_http_done: called premature == 1
* stopped the pause stream!
* Closing connection 0
* TLSv1.2 (OUT), TLS alert, Client hello (1):
curl: (51) SSL: no alternative certificate subject name matches target host name '10.103.37.154'
```

So I need to regenerate that cert, since I'm not using the hostname `docker-registry` anymore :/

geerlingguy, 2018-06-26:
Once I got the right hostname and added it in /etc/hosts, I get:

```
root@kube1:/home/vagrant# curl -vvvv https://registry.pidramble.test/debug/health
*   Trying 127.0.0.1...
* TCP_NODELAY set
* connect to 127.0.0.1 port 443 failed: Connection refused
* Failed to connect to registry.pidramble.test port 443: Connection refused
* Closing connection 0
curl: (7) Failed to connect to registry.pidramble.test port 443: Connection refused
```

So at least it's externally and internally consistent now :)

geerlingguy, 2018-07-18:
So close on this :P

geerlingguy, 2018-07-28:
It works! I went with an insecure HTTP registry, which is not at all the end goal, but I'll open two follow-up issues for making it more secure.

closed 2018-07-28 by geerlingguy

MysticalMount, 2020-03-02:
Everywhere I go I see geerlingguy, from building my Vagrant base builds on VirtualBox for export to my KVM hypervisor, through to deploying Docker to my home Kubernetes cluster, keep on at it dude your stuff was very helpful to me in my home projects!

geerlingguy, 2020-03-03:
@MysticalMount - Haha, well I'm glad that I could help at all those points. And I guess if you keep running into my work, I approve of the tech you're using :D

commit 79e715d5 (2018-07-28): Fixes #108: Deploy a local Docker registry inside Kubernetes cluster.