geerlingguy/raspberry-pi-dramble #109: Don't make MySQL a Pet

geerlingguy, 2018-06-06:
Right now, the simple way of getting some sort of reliability for MySQL data is to use a PersistentVolume stored locally on the `kube5` node, using `nodeAffinity` to tell the PV where to go. There are a few other ways to do things sort-of-simply, and we could also use an NFS volume for the MySQL data... but that would be insane and slow).

But the holy grail would be some sort of MySQL replication setup where:

  - MySQL doesn't care what node it's running on.
  - MySQL doesn't rely on instance-based storage (so is not tied to a particular instance).
  - Drupal doesn't care about masters and replicas or particular nodes; it just points at a MySQL endpoint.
  - You can delete the MySQL master pod/container and there might be a minute or two of downtime, but a new master pops up in its place.

I was intrigued by this [MySQL on Autopilot](https://www.joyent.com/blog/dbaas-simplicity-no-lock-in) blog post, specifically:

> The autopilot pattern is an approach to application and infrastructure design that pushes automation for each component of our systems into the application. Each container that makes up the application has its own lifecycle, and we package those lifecycle behaviors into the application container rather than relying on external infrastructure.

See also: https://github.com/autopilotpattern/mysql

geerlingguy, 2018-06-06:
It might be too complex to handle inside this particular project... but having this project around means I can tinker with getting the pattern working inside Kubernetes.

onedr0p, 2019-07-09:
You should take a look at spinning up a GlusterFS cluster for persistent data. It even has an API for Kubernetes called [heketi](https://github.com/heketi/heketi).

I have a 3 node Pi4 cluster set up and it's pretty neat.

Your containers can be HA, why not the data :)