geerlingguy/raspberry-pi-dramble #136: Where is the playboooks located?

poon64, 2019-01-02:
In the documentation it's referring to the playbooks directory, but it can't be found in the repository. 
Like this, 
 ansible-galaxy install -r playbooks/requirements.yml --force
ansible-playbook -i inventory playbooks/drupal/main.yml

geerlingguy, 2019-01-04:
@poon64 - If you're using the latest version of the Dramble cluster (which uses Kubernetes), there is no more `playbooks` directory; as mentioned in the README (https://github.com/geerlingguy/raspberry-pi-dramble#setting-up-the-pis), a couple of the pages on the Wiki still need to be updated, as I'm working quite a bit on making sure the Kubernetes setup process is working well before I complete the documentation edits.

geerlingguy, 2019-01-04:
See related: #133

closed 2019-01-04 by geerlingguy