geerlingguy/raspberry-pi-dramble #143: Build scripts (and/or just a user guide) for demo of Ansible + Blinksticks

geerlingguy, 2019-02-20:
From #112, I've broken out this issue:

* [ ] Build a few scripts that allow easy remote control of LEDs (for use with demoing Ansible ad-hoc/playbook orchestration).

Basically, I want to be able to do some slick things in a demo like:

# Stop the `dramble-node-monitor` service so the LEDs go off and the Pis are not monitoring node status anymore.
# Turn the blinkstick blue, using serial=1
# Turn the blinkstick green, using serial=2
# Turn the blinkstick off, with no serial (so all at once)

It would also be sorta cool if I built a sidecar pod to the Drupal pod that would flash the blinkstick any time a request was made to that container (e.g. watch the Apache log?)... but that would probably be something to break out into a separate feature.

I would also like to document the setup and use of blinksticks (vs soldered LED boards) in the Dramble Wiki.

geerlingguy, 2019-03-11:
See: https://github.com/geerlingguy/raspberry-pi-dramble/blob/master/testing/presentation/SCRIPT.md

closed 2019-03-11 by geerlingguy

SamTheFishy, 2019-06-18:
Thanks very useful.