geerlingguy/raspberry-pi-dramble #147: Take a look at K3s - lightweight Kubernetes-certified distro

geerlingguy, 2019-02-27:
See: https://k3s.io

* "Both ARM64 and ARMv7 are supported with binaries and multiarch images available for both. k3s works great from something as small as a Raspberry Pi or as large as an AWS a1.4xlarge 32GiB server."
* "k3s is wrapped in a simple package that reduces the dependencies and steps needed to run a production Kubernetes cluster. Packaged as a single binary, k3s makes installation and upgrade as simple as copying a file. TLS certificates are automatically generated to ensure that all communication is secure by default."
* "K3s is a Certified Kubernetes distribution designed for production workloads in unattended, resource-constrained, remote locations or inside IoT appliances."

jkupcho, 2020-03-25:
k3s is dead simple to spin up, and provides some really nice to haves out of the box.
* an easy to use / understand local storage PVC
* traefik installed by default for ingress (though, should probably upgrade to v2, stable helm release uses v1, I removed and reinstalled v2)
* really easy LoadBalancer, if the Service exposes a port, it's exposed through that node's host port (again, should probably handle through traefik)

rgarrigue, 2020-04-01:
There are roles for k3s here https://github.com/rancher/k3s/tree/master/contrib/ansible