geerlingguy/raspberry-pi-dramble #150: Configure Flannel networking task failure

457r0ch1mp, 2019-03-24:
Hi Jeff,

The geerlingguy.kubernetes module is failing with the following error:

```
failed: [kube1] (item=kubectl apply -f https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml) => changed=false
  cmd:
  - kubectl
  - apply
  - -f
  - https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml
  delta: '0:00:01.989797'
  end: '2019-03-24 05:11:36.882905'
  item: kubectl apply -f https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml
  msg: non-zero return code
  rc: 1
  start: '2019-03-24 05:11:34.893108'
  stderr: |-
    unable to recognize "https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml": Get https://192.168.1.101:6443/api?timeout=32s: dial tcp 192.168.1.101:6443: connect: connection refused
    unable to recognize "https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml": Get https://192.168.1.101:6443/api?timeout=32s: dial tcp 192.168.1.101:6443: connect: connection refused
  stderr_lines:
  - 'unable to recognize "https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml": Get https://192.168.1.101:6443/api?timeout=32s: dial tcp 192.168.1.101:6443: connect: connection refused'
  - 'unable to recognize "https://raw.githubusercontent.com/coreos/flannel/master/Documentation/k8s-manifests/kube-flannel-rbac.yml": Get https://192.168.1.101:6443/api?timeout=32s: dial tcp 192.168.1.101:6443: connect: connection refused'
  stdout: ''
  stdout_lines: <omitted>
failed: [kube1] (item=kubectl apply -f https://raw.githubusercontent.com/coreos/flannel/master/Documentation/kube-flannel.yml) => changed=false
  cmd:
  - kubectl
  - apply
  - -f
  - https://raw.githubusercontent.com/coreos/flannel/master/Documentation/kube-flannel.yml
  delta: '0:00:01.749090'
  end: '2019-03-24 05:11:39.570509'
  item: kubectl apply -f https://raw.githubusercontent.com/coreos/flannel/master/Documentation/kube-flannel.yml
  msg: non-zero return code
  rc: 1
  start: '2019-03-24 05:11:37.821419'
  stderr: |-
    unable to recognize "https://raw.githubusercontent.com/coreos/flannel/master/Documentation/kube-flannel.yml": Get https://192.168.1.101:6443/api?timeout=32s: dial tcp 192.168.1.101:6443: connect: connection refused
```

I think this might be related to https://github.com/coreos/flannel/issues/1031#issue-353485708

Is there a work-around that can get me a running dramble to work with?

Thanks!

geerlingguy, 2019-03-25:
Hmm, it seems like in this case the manifest itself isn’t being applied correctly. I haven’t seen this issue recently, but I wonder if you downloaded the manifest manually and then kubectl apply’ed it, if that would work.

geerlingguy, 2019-03-28:
Just saw this same failure in the automated test build of my Kubernetes role, so I have an upstream issue there: https://github.com/geerlingguy/ansible-role-kubernetes/issues/42

geerlingguy, 2019-04-19:
Testing a fix via https://github.com/geerlingguy/ansible-role-kubernetes/pull/46

geerlingguy, 2019-04-19:
Now hitting: https://github.com/kubernetes/kubernetes/issues/75683

commit 49d07c97 (2019-04-19): Fixes #150: Update to Kubernetes 1.13.5 so everything works.
closed 2019-04-19 by geerlingguy