geerlingguy/raspberry-pi-dramble #158: What's a Gramble?

onedr0p, 2019-07-11:
Not sure how to notify you other than this issue but I am working on a project that hopefully can tie into yours. Instead of using NFS as a datastore for Kubernetes I am working on automating setting up a GlusterFS cluster. It use Heketi so k8s can communicate with it's RESTful API to create volumes.

Still kinda a WIP but here is the sauce. A lot is borrowed from your project, even the name 👍 

https://github.com/onedr0p/raspberry-pi-gramble

If you would like gramble could be migrated to your namespace, with me added as a maintainer. It's up to you. Anyways, thanks again for your project, you have inspired me :)

geerlingguy, 2019-07-17:
Ha! Sounds interesting. The first version of this cluster, waaay back when, actually used GlusterFS instead of NFS, but I had a lot of maintenance headaches which didn't occur quite as often with NFS, so NFS has kinda stuck.

See: https://github.com/geerlingguy/raspberry-pi-dramble/issues/55

I'm happy to see forks of the dramble, and am glad you got something out of it!

closed 2019-07-17 by geerlingguy

onedr0p, 2019-07-17:
Might be worth revisiting, 4 years is a long time!

Heketi is really awesome with Kubernetes.