geerlingguy/raspberry-pi-dramble #161: Add two more microSD cards to the benchmarks

geerlingguy, 2019-07-17:
After the last round of benchmarks, I purchased two new microSD cards that I would like to test:

  1. [SanDisk Extreme 64GB U3 V30 A2](https://www.amazon.com/gp/product/B07FCMBLV6/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=d65c4ec6f33497b1596134e3ac06da2a&language=en_US)
  1. [Samsung Pro Endurance 32GB U1](https://www.amazon.com/gp/product/B07B98GXQT/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=6bec0a6dbb665304e450c086e2b704fc&language=en_US)

For the former, some people suggested A2 cards may have better random I/O characteristics. For the latter, many suggested these cards can handle multiple write cycles better (though I wonder if it's more optimized for the 'write entire card then write over entire card' use case of a dashcam, than for random small file writes...).

geerlingguy, 2019-07-17:
## SanDisk Extreme 64GB U3 V30 A2

  - hdparm: 43.73 MB/sec
  - dd write: 36.3 MB/sec
  - 4K read: 7.35 MB/sec
  - 4K write: 3.70 MB/sec

## Samsung Pro Endurance 32GB U1

  - hdparm: TODO MB/sec
  - dd write: TODO MB/sec
  - 4K read: TODO
  - 4K write: TODO

geerlingguy, 2019-07-18:
I'm kind of amazed that the A2 is getting _worse_ random I/O / IOPS on the Pi than the 'super old and totally outdated' A1/non-A-class Sandisk Extreme... sounds like either I have a terrible card, the Pi has a terrible microSD interface, or it's marketing wank.

geerlingguy, 2019-07-18:
To test the theory above, I used [AmorphousDiskMark](http://www.katsurashareware.com/pgs/adm.html) using a brand new UHS-II XQD/SD USB 3.0 reader on my Mac, with the A2 microSD card formatted as FAT32. I couldn't use `iozone` because it gets caught up in macOS's built-in filesystem caching, which makes the numbers go a few orders of magnitude off from reality.

Running 4K I/O benchmarks now.

geerlingguy, 2019-07-18:
Using ADM:

  - 4K read: 6.81 MB/sec
  - 4K write: TODO MB/sec

geerlingguy, 2019-07-18:
I'm not satisfied condemning the 'A2' designation based on one card, so I've also ordered a [Lexar Professional 128GB microSDXC UHS-I/U3 A2 V30](https://www.amazon.com/gp/product/B07NHCB6N3/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=6835091df3496d8b362eb7893a175bf0&language=en_US) card. I'm waiting for it to arrive, might take another day or two.

geerlingguy, 2019-07-20:
## Lexar Professional U3 A2 V30 128GB

  - hdparm: 44.35 MB/sec
  - dd write: 32.4 MB/sec
  - 4K read: 9.65 MB/sec (2470 IOPS)
  - 4K write: 2.55 MB/sec (653 IOPS)

geerlingguy, 2019-07-20:
Yeah... so A2 means absolutely nothing. Completely made-up marking BS to sell 2x more expensive cards.

I also tested on another brand new microSD and SD card USB 3.1 Type C reader, which supports UHS-II for SD (doesn't mean anything for microSD AFAICT though), and it was giving similar results.

geerlingguy, 2019-07-20:
Also taking a glance through the [SD 6.0 Part I physical specification](https://www.sdcard.org/downloads/pls/index.html) to see if there's something huge I'm missing here...

geerlingguy, 2019-07-20:
There's nothing else indicating what kind of conditions would be different, however I have been testing with a 100MB file/block size in the past, so I'm going to run IOzone with a 256MB file to see if it makes any difference:

    ./iozone -e -I -a -s 256M -r 4k -i 0 -i 2

geerlingguy, 2019-07-20:
Nope, still 9.58 MB/sec read, 2.55 MB/sec write. Weak. Results in all cases differ < 1%... at least the card is very reliable.

geerlingguy, 2019-07-21:
A chart, to finish out this issue:

[image]

It saddens me the SD Association can get away with this.

closed 2020-04-24 by stale[bot]

valenroy, 2020-06-06:
Have you tested cards from 128GB and above? Generally the marketed performance indicators are based off the cards from those capacities and above. If you compare the performance matrix of all the available storage capacities, the results vary between 32GB, 64GB and 128GB. The performance seems to plateau from 128GB upwards. It would be interesting if you can further investigate this with A2 specified cards.

geerlingguy, 2020-06-07:
@valenroy - I have done a number of tests with 128 GB cards.

The performance for large file (e.g. 500+ MB files) sequential copy is slightly better (very slight, no real-world difference IMO) for larger cards, but the random access is pretty much exactly the same (within 1% margin of error) for all sizes. The sad fact is most cards are made to perform well for large file activity, not for random access.