geerlingguy/raspberry-pi-dramble #167: Service Unavailable when using testing/vagrant cluster build

mk53202, 2019-11-30:
I'm receiving a 'Service Unavailable' when running the vagrant cluster build.
I'm accessing http://cluster.pidramble.test/ and I ssh into the controller and everything in the cluster is up:

root@kube1:/home/vagrant# kubectl get nodes
NAME    STATUS   ROLES    AGE   VERSION
kube1   Ready    master   44m   v1.15.0
kube2   Ready    <none>   43m   v1.15.0
kube3   Ready    <none>   43m   v1.15.0
kube4   Ready    <none>   43m   v1.15.0

Any ideas on how to troubleshooting this more?

buckaroogeek, 2019-11-30:
I had a similar problem yesterday. Vagrant up started all VMs and then returned to my workstation prompt.  And nothing was listening on port 80 at cluster.piramble.test (ping would work). In hind sight it appears that Ansible for some reason was not provisioning the VMs. Not sure what was the solution but I did a vagrant destroy to clear out all VMs and ran vagrant up. This time I started to see some ansible errors complaining about missing roles. I would manually install the missing role then run vagrant provision, encounter another missing role, and repeat. After installing about 6 roles, Ansible started to execute a long series of Tasks which properly configured the system.

mk53202, 2019-11-30:
> I had a similar problem yesterday. Vagrant up started all VMs and then returned to my workstation prompt. And nothing was listening on port 80 at cluster.piramble.test (ping would work). In hind [...]

I ran into the missing roles too (security role, etc...) but got past that by installing the requirements.txt. All the ansible stuff is running with zero errors. I'm still having the service unavailable. Port 80 is up but it doesn't appear to be forwarding request right. Not sure what else I'm missing.

mk53202, 2019-11-30:
I'm stumped because everything looks up.

buckaroogeek, 2019-11-30:
Since i do not routinely run ansible, I  am not sure I can help. I ran `ansible-galaxy install -r requirements.yml --force` in the root directory of the project and `ansible-galaxy install -r ../../requirements.yml --force` in the ./testing/vagrant directory. One or the other seems to have done the trick for me. I am using a fedora 30 workstation.

geerlingguy, 2019-12-06:
@mk53202 / @buckaroogeek - I just ran this a few days back (on macOS Catalina) and didn't seem to have an issue, so if it's giving Service Unavailable, then it could be routing, or maybe a hosts file issue. Did you make sure that `ping cluster.piramble.test` worked?

buckaroogeek, 2019-12-07:
@geerlingguy - Greetings. I can confirm that the ping did work. My suspicion is that the service unavailable message was related somehow to missing roles. This is now working for me but will try a fresh install later this weekend.

closed 2020-04-05 by stale[bot]