geerlingguy/raspberry-pi-dramble #169: Undefined variable, The error was: 'dict object'

purencool, 2020-01-02:
Hi,

Great work firstly, I don't know is this a bug or I am not following you howto properly. So I have created this ticket to see which one it is.  I have Two PI's 4, 2gig with a Debian installation.

Testing your configuration it works fine:

```
ansible all -i inventory -m ping
[WARNING]: Platform linux on host 192.168.20.17 is using the discovered Python interpreter at /usr/bin/python, but future installation of
another Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more
information.

192.168.20.17 | SUCCESS => {
    "ansible_facts": {
        "discovered_interpreter_python": "/usr/bin/python"
    },
    "changed": false,
    "ping": "pong"
}
[WARNING]: Platform linux on host 192.168.20.18 is using the discovered Python interpreter at /usr/bin/python, but future installation of
another Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more
information.

192.168.20.18 | SUCCESS => {
    "ansible_facts": {
        "discovered_interpreter_python": "/usr/bin/python"
    },
    "changed": false,
    "ping": "pong"
}
Sitecore:networking john.cullen$  ansible all -i inventory -a "df -h"
[WARNING]: Platform linux on host 192.168.20.18 is using the discovered Python interpreter at /usr/bin/python, but future installation of
another Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more
information.

192.168.20.18 | CHANGED | rc=0 >>
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  1.3G   27G   5% /
devtmpfs        841M     0  841M   0% /dev
tmpfs           970M     0  970M   0% /dev/shm
tmpfs           970M  8.4M  962M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           970M     0  970M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   52M  201M  21% /boot
tmpfs           194M     0  194M   0% /run/user/1000

[WARNING]: Platform linux on host 192.168.20.17 is using the discovered Python interpreter at /usr/bin/python, but future installation of
another Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more
information.

192.168.20.17 | CHANGED | rc=0 >>
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  1.3G   27G   5% /
devtmpfs        841M     0  841M   0% /dev
tmpfs           970M     0  970M   0% /dev/shm
tmpfs           970M  8.5M  962M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           970M     0  970M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   52M  201M  21% /boot
tmpfs           194M     0  194M   0% /run/user/1000

ansible all -i inventory -a "free -m"
[WARNING]: Platform linux on host 192.168.20.17 is using the discovered Python interpreter at /usr/bin/python, but future installation of
another Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more
information.

192.168.20.17 | CHANGED | rc=0 >>
              total        used        free      shared  buff/cache   available
Mem:           1939          70        1782           8          86        1784
Swap:            99           0          99

[WARNING]: Platform linux on host 192.168.20.18 is using the discovered Python interpreter at /usr/bin/python, but future installation of
another Python interpreter could change this. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html for more
information.

192.168.20.18 | CHANGED | rc=0 >>
              total        used        free      shared  buff/cache   available
Mem:           1939          61        1749           8         128        1791

```

Then I navigate to the following directory
`raspberry-pi-dramble/setup/networking`

and set up my variables in `vars.yml`

```
---
# Mapping of what hardware MAC addresses should be configured with specific IPs.
mac_address_mapping:
  "macaddressofmachine":
    name: k1.purencool.net
    ip: "192.168.20.45"
  "macaddressofmachine":
    name: k2.purencool.net
    ip: "192.168.20.46"

# Nameservers to use in resolv.conf.
dns_nameservers:
  - "8.8.8.8"
  - "8.8.4.4"

```

My inventory file looks like this

```
[dramble]
192.168.20.17
192.168.20.18

[dramble:vars]
ansible_ssh_user=pi

```

When I try the following command `ansible-playbook -i inventory main.yml`. I get the following output

```

PLAY [dramble] **********************************************************************************************************************************

TASK [Gathering Facts] **************************************************************************************************************************
ok: [192.168.20.18]
ok: [192.168.20.17]

TASK [Set the current MAC address for eth0.] ****************************************************************************************************
ok: [192.168.20.17]
ok: [192.168.20.18]

TASK [Set variables based on eth0 MAC address.] *************************************************************************************************
fatal: [192.168.20.17]: FAILED! => {"msg": "The task includes an option with an undefined variable. The error was: 'dict object' has no attribute 'mac-address-i-have-removed'\n\nThe error appears to be in '/path-I-have-removed/raspberry-pi-dramble/setup/networking/main.yml': line 13, column 7, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n    - name: Set variables based on eth0 MAC address.\n      ^ here\n"}
fatal: [192.168.20.18]: FAILED! => {"msg": "The task includes an option with an undefined variable. The error was: 'dict object' has no attribute 'mac-address-i-have-removed'\n\nThe error appears to be in '/path-I-have-removed/raspberry-pi-dramble/setup/networking/main.yml': line 13, column 7, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n    - name: Set variables based on eth0 MAC address.\n      ^ here\n"}

PLAY RECAP **************************************************************************************************************************************
192.168.20.17              : ok=2    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0   
192.168.20.18              : ok=2    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0  

```

This is line 13 `  - name: Set variables based on eth0 MAC address.`

This is the main.yml

```

---
- hosts: dramble
  become: yes

  vars_files:
    - vars.yml

  tasks:
    - name: Set the current MAC address for eth0.
      set_fact:
        dramble_mac_address: "{{ hostvars[inventory_hostname].ansible_default_ipv4.macaddress }}"

    - name: Set variables based on eth0 MAC address.
      set_fact:
        dramble_hostname: "{{ mac_address_mapping[dramble_mac_address].name }}"
        dramble_ip_address: "{{ mac_address_mapping[dramble_mac_address].ip }}"

    - name: Set up networking-related files.
      template:
        src: "templates/{{ item.template }}"
        dest: "{{ item.dest }}"
        mode: 0644
      with_items:
        - { template: hostname.j2, dest: /etc/hostname }
        - { template: hosts.j2, dest: /etc/hosts }
        - { template: resolv.conf.j2, dest: /etc/resolv.conf }
        - { template: dhcpcd.conf.j2, dest: /etc/dhcpcd.conf }
      notify:
        - update hostname
        - delete dhcp leases

  handlers:
    - name: update hostname
      command: "hostname {{ dramble_hostname }}"

    - name: delete dhcp leases
      file:
        path: /var/lib/dhcp/dhclient.leases
        state: absent
      with_items:
        - /var/lib/dhcp/dhclient.leases
        - /var/lib/dhcpcd5/dhcpcd-eth0.lease

```

My question is: I am not sure what it is that creating the error are you able to see what it is?

amajot, 2020-01-05:
@purencool I had this same error, the issue was that the characters in my MAC addresses were uppercase (copy/pasted from my router). Ansible is expecting lowercase. If you look at the error message it shows the MAC as lowercase, which is the clue for what the problem is

purencool, 2020-01-05:
@amajot, yes that was the issue, the MAC addresses needed to be lowercase in circumstance. Thanks for the information.

closed 2020-01-05 by purencool

geerlingguy, 2020-01-10:
Thanks for posting the fix!