geerlingguy/raspberry-pi-dramble #170: Endpoints not available for drupal/drupal

CodeDestroyer, 2020-01-28:
I am running on raspbian 2019-09-26 with aarch64 enabled.

I have run through everything, and it all seems to standup.  
The only change was the private network range

when i try to hit the master 192.168.2.2 with host cluster.pidramble.test it cannot connect.
if i try any of the other pods i get a 404.

looking into the kubernetes logs i am seeing `describe pod/traefik-ingress-controller-4ndhw -n kube-system`
```
level=info msg="Server configuration reloaded on :443"
time="2020-01-28T20:17:15Z" level=info msg="Server configuration reloaded on :80"
time="2020-01-28T20:17:15Z" level=info msg="Server configuration reloaded on :8080"
time="2020-01-28T20:17:15Z" level=warning msg="Endpoints not available for drupal/drupal"
time="2020-01-28T19:54:48Z" level=warning msg="Endpoints not available for registry/docker-registry"```

mikev1963, 2020-03-30:
I have the same issue. Why is NodePort not using port 80?  Here is my issues below:
```
$ kubectl get svc -n nginx
NAME    TYPE       CLUSTER-IP      EXTERNAL-IP   PORT(S)        AGE
nginx   NodePort   10.111.23.204   <none>        80:32357/TCP   4m31s
pi@master:/mnt/yaml $ 
pi@master:/mnt/yaml $ curl --head http://master.mikevware.net
curl: (7) Failed to connect to master.mikevware.net port 80: Connection refused
$ 
$ curl --head http://master.mikevware.net:32357
HTTP/1.1 200 OK
Server: nginx/1.17.9
Date: Mon, 30 Mar 2020 15:54:43 GMT
Content-Type: text/html
Content-Length: 612
Last-Modified: Tue, 03 Mar 2020 14:32:47 GMT
Connection: keep-alive
ETag: "5e5e6a8f-264"
Accept-Ranges: bytes
```

mikev1963, 2020-03-31:
Jeff.  Are you forwarding your website to the ingress port ?  Since port 80 is not available to my domain.  Should I forward my website to port 32357 as shown in the above post?

geerlingguy, 2020-06-07:
Could there be some other service running on port 80 on your Pis? If so, Traefik would not be able to bind to the port and serve traffic for Drupal there.

closed 2020-10-05 by stale[bot]