geerlingguy/raspberry-pi-dramble #178: main.yml hangs when installing [geerlingguy.pip : Ensure pip_install_packages are installed.]

NiftyMist, 2020-05-23:
I'm trying this out on a 5 node cluster of raspberry pis.  They are all fresh installs of 2020-02-13-raspbian-buster-lite.img.  I was originally getting fails when getting to this task.  I added `ansible_python_interpreter=/usr/bin/python3` to my dramble vars in the inventory and now the task just hangs until I cancel the playbook.  Is there any workaround for this or something I'm missing?

```
TASK [geerlingguy.pip : Ensure pip_install_packages are installed.] *************************
[DEPRECATION WARNING]: Invoking "pip" only once while using a loop via squash_actions is 
deprecated. Instead of using a loop to supply multiple items and specifying `name: "{{ 
item.name | default(item) }}"`, please use `name: '{{ pip_install_packages }}'` and remove 
the loop. This feature will be removed in version 2.11. Deprecation warnings can be disabled
 by setting deprecation_warnings=False in ansible.cfg.
[DEPRECATION WARNING]: Invoking "pip" only once while using a loop via squash_actions is 
deprecated. Instead of using a loop to supply multiple items and specifying `name: "{{ 
item.name | default(item) }}"`, please use `name: '{{ pip_install_packages }}'` and remove 
the loop. This feature will be removed in version 2.11. Deprecation warnings can be disabled
 by setting deprecation_warnings=False in ansible.cfg.
[DEPRECATION WARNING]: Invoking "pip" only once while using a loop via squash_actions is 
deprecated. Instead of using a loop to supply multiple items and specifying `name: "{{ 
item.name | default(item) }}"`, please use `name: '{{ pip_install_packages }}'` and remove 
the loop. This feature will be removed in version 2.11. Deprecation warnings can be disabled
 by setting deprecation_warnings=False in ansible.cfg.
[DEPRECATION WARNING]: Invoking "pip" only once while using a loop via squash_actions is 
deprecated. Instead of using a loop to supply multiple items and specifying `name: "{{ 
item.name | default(item) }}"`, please use `name: '{{ pip_install_packages }}'` and remove 
the loop. This feature will be removed in version 2.11. Deprecation warnings can be disabled
 by setting deprecation_warnings=False in ansible.cfg.
[DEPRECATION WARNING]: Invoking "pip" only once while using a loop via squash_actions is 
deprecated. Instead of using a loop to supply multiple items and specifying `name: "{{ 
item.name | default(item) }}"`, please use `name: '{{ pip_install_packages }}'` and remove 
the loop. This feature will be removed in version 2.11. Deprecation warnings can be disabled
 by setting deprecation_warnings=False in ansible.cfg.
^C [ERROR]: User interrupted execution

```

geerlingguy, 2020-05-24:
@NiftyMist - That task can take a _really_ long time, especially on older Pis. The Pi 4 can take a few minutes. The Pi 3 B+ a bit longer. The Pi 2 is extremely slow, and it could take 20+ minutes there.

If you want to see if it's actually working, log into one of the servers and run `top` and see if `pip3` or `python3` are using a lot of CPU.

NiftyMist, 2020-05-26:
@geerlingguy I believe this issue was name resolution error on pypi's side.  I have since be able to get the openshift pip3 package installed.  However now the playbook is failing on the flannel networking install tasks.  I'm looking into issue.

geerlingguy, 2020-05-26:
In that case I'll close this issue since it seems like it was a temporary pypi issue. Please open a new issue if you find something with the flannel config that needs correcting.

closed 2020-05-26 by geerlingguy