geerlingguy/raspberry-pi-dramble #183: Add 'USB benchmarks' page

geerlingguy, 2020-07-11:
I have been doing a number of USB benchmarks recently, and finally with the Pi 4, the included 2x USB 3.0 ports offer a _massive_ improvement—in many cases—over using microSD storage. In the past the USB 2.0 ports on the Pi 3, 3+, 2, etc. were severely hampered by no UASP support, 480 Mbps top bandwidth, and sharing a bus with the network jack.

So I'd like a comprehensive overview of USB 3.0 performance on the Pi 4 with a variety of devices.

The contenders:

| Product | USB 3.0 Support? | UASP (on Pi 4)? | Price (Total) |
| --- | --- | --- | --- |
| [Inateck SATA enclosure](https://www.amazon.com/gp/product/B00FCLG65U/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=0491814fd47076524c84c853bc68d451&language=en_US) + [Kinsgston 240 GB SSD](https://www.amazon.com/gp/product/B01N5IB20Q/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=19b8b148502d329914f33849d9752474&language=en_US) | Yes | Yes | $48.98
| [TDBT m.2 NVMe Enclosure](https://www.amazon.com/TDBT-Enclosure-Thermal-Cooling-External/dp/B07TJT6W8K/ref=as_li_ss_tl?crid=M5RP10LQV7P4&dchild=1&keywords=tdbt+m.2+nvme+ssd+enclosure+with+heatsink&qid=1594428301&s=electronics&sprefix=tdbt+m.2+nvme,electronics,157&sr=1-3&linkCode=ll1&tag=mmjjg-20&linkId=08ef39c2a221af0334de67928940c938&language=en_US) + [XPG SX6000 Lite 128 GB](https://www.amazon.com/XPG-SX6000-Gen3x4-1200MB-ASX6000LNP-128GT-C/dp/B07MTVVXG7/ref=as_li_ss_tl?dchild=1&keywords=xpg+sx6000+3d+nand+128&qid=1594428327&s=electronics&sr=1-6&linkCode=ll1&tag=mmjjg-20&linkId=749b4faec1a306f7668a3ffb1ecdab1c&language=en_US) | Yes | TODO | $64.64 |
| [Arcanite 128 GB USB 3.1 flash drive](https://www.amazon.com/ARCANITE-128GB-USB-Flash-Drive/dp/B07RWPF43G/ref=as_li_ss_tl?dchild=1&keywords=arcanite+128gb&qid=1594428286&s=electronics&sr=1-2&linkCode=ll1&tag=mmjjg-20&linkId=e578ecbd7e284e1ebe42313ca3c40118&language=en_US) | Yes | No | $23.99 |
| [SanDisk Ultra Fit 128 GB flash drive](https://www.amazon.com/gp/product/B01BGTG2A0/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=4cb69b182511914eb6368955506ab460&language=en_US) | Yes | TODO | $21.99 |
| [SanDisk Ultra Flair 16 GB flash drive](https://www.amazon.com/gp/product/B015CH1GTO/ref=as_li_ss_tl?ie=UTF8&psc=1&linkCode=ll1&tag=mmjjg-20&linkId=0aceb0d14d2ebb193932d0d3350bd239&language=en_US) | USB 3.0 | TODO | $6.69 |
| [Samsung 32GB Evo+ microSD](https://www.amazon.com/Samsung-256GB-microSDXC-Adapter-MB-MC256GA/dp/B00WR4IJBE/ref=as_li_ss_tl?dchild=1&keywords=micro+sd&qid=1594428649&sr=8-21&th=1&linkCode=ll1&tag=mmjjg-20&linkId=cdf742dfcd473427cc9c987096bee1da&language=en_US) | No | No | $9.10 |

geerlingguy, 2020-07-11:
## Benchmarks

TODO.

geerlingguy, 2020-07-11:
## Results

### On Raspberry Pi 4, USB 3.0:

| Product | `hdparm` | `dd` | 4K read | 4K write |
| --- | --- | --- | --- | --- |
| Inateck + Kingston SSD | TODO MB/s | TODO MB/s | TODO MB/s | TODO MB/s |
| TDBT + XPG NVMe | 337.59 MB/s | 217.33 MB/s | 22.12 MB/s | 34.23 MB/s |
| Arcanite Flash Drive | 323.32 MB/s | 176 MB/s | 9.68 MB/s | 3.72 MB/s |
| Sandisk Ultra Fit | TODO MB/s | TODO MB/s | TODO MB/s | TODO MB/s |
| Sandisk Ultra Flair | TODO MB/s | TODO MB/s | TODO MB/s | TODO MB/s |
| Samsung Evo+ microSD | 48.83 MB/s | 34.70 MB/s | 10.43 MB/s | 3.72 MB/s |

### On MacBook Pro, ThunderBolt 3:

| Product | `hdparm` | `dd` | 4K read | 4K write |
| --- | --- | --- | --- | --- |
| Inateck + Kingston SSD | TODO | TODO | TODO | TODO |
| TDBT + XPG NVMe | N/A | 351.82 MB/s | 2030.90 MB/s | 232.08 MB/s |
| Arcanite Flash Drive | TODO | TODO | TODO | TODO |
| Sandisk Ultra Fit | TODO | TODO | TODO | TODO |
| Sandisk Ultra Flair | TODO | TODO | TODO | TODO |
| Samsung Evo+ microSD | TODO | TODO | TODO | TODO |

geerlingguy, 2020-07-13:
A quick note comparing the worst versus the best in terms of consistency—a lot of people might not think as much about this, but the thermals are incredibly important. When parts get way too hot and they throttle, you leave a lot of performance on the table.

Here's the ultra fit doing a stress run:

[image]

And here's the Corsair GTX:

[image]

At least 4x better performance, and half the heat due to the case basically being a massive heat sink.

faceless2, 2021-07-29:
Jeff, thanks for the code. I don't know if you're collecting these, but if so: from 4 runs on Pi 4, discarded first, averaged:

Product | hdparm | dd | 4K read | 4K write
-------- | ------- | ---- | -------- | --------
Kingston DataTraveler G4 32MB | 148 MB/s|15.7 MB/s|9.3 MB/s|5.8 MB/s