geerlingguy/raspberry-pi-dramble #184: Trouble installing openshift

jbarb365, 2020-09-05:
I am encountering an error while provisioning the dramble cluster.
**ansible-playbook -i inventory main.yml 

The following error appears for each kube in the cluster: 

An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ImportError: No module named pkg_resources
fai
[dramble_setup_output.pdf](https://github.com/geerlingguy/raspberry-pi-dramble/files/5178353/dramble_setup_output.pdf)
led: [kube4] (item=['openshift']) => changed=false 
  ansible_loop_var: item
  item:
  - openshift
  msg: Failed to import the required Python library (setuptools) on kube4's Python /usr/bin/python. Please read module documentation and install in the appropriate location. If the required library is installed, but Ansible is using the wrong Python interpreter, please consult the documentation on ansible_python_interpreter

outlawstar4761, 2020-09-10:
I just ran into this experience myself. First I tried `pip install openshift` on each pi and re-ran the playbook. I noticed a complaint about my PATH variable but ignored it. Rerunning the playbook, I got the same results. I returned to each pi and executed `sudo pip install openshift` and reran the playbook successfully. I didn't fully test this theory but I'm working off the assumption that installing pip libraries as root adds them to PATH automatically and that updating PATH after the install as pi may have resolved the issue.

tobiasweede, 2020-10-07:
Faced the same issue on a RPi.

My workaround was to adjust the softlink to the default python interpreter
`ln -sf /usr/bin/python3 /usr/bin/python`
It pointed to python2 by default...

Could probably also be resolved by via
`ansible_python_interpreter=/usr/bin/python3`
e.g. in ansible.cfg but didn't test it that way.

closed 2021-02-05 by stale[bot]