geerlingguy/raspberry-pi-dramble #189: Does Pi-Dramble work with PI4's?

garylefebvreg, 2021-01-03:
Hello, I have been working on the Pi-dramble and I am using raspberry pi 4's. The OS i am using is Debian Raspberry Pi OS and can not get ansible to setup the netwoks. I know you suggest using Raspian Lite, but it is a few years old. I don't think that it works well with PI4's. Is there something I am missing or is this project for the archives...

kind regards,

geerlingguy, 2021-01-03:
This project is made to run well on the Raspberry Pi 4 (the current setup uses four Raspberry Pi 4s). The last time I set it up Raspberry Pi OS was not yet released (it was still called 'Raspbian'), but I have since started moving everything I do to the 64-bit Pi OS beta.

But everything should work (with potentially a couple tiny modifications) on either 64-bit or 32-bit OSes, and should even work on Ubuntu for Raspberry Pi.

Note that I haven't set up a new Dramble in the past 6 months or so, so there's a chance one or two of the automation scripts in this repository has a bug that requires an update.

garylefebvreg, 2021-01-04:
Yes, thank you! I gave it a couple of tries and I am getting there! I
really enjoy this project. I have the Ansible for Dev Ops . Looking
forward!!! Thank you!!!

garylefebvreg, 2021-01-10:
Hello again, I can not get "pip install openshift" to install on my Macos. The errors are concerning the 

Building wheels for collected packages: ruamel.ordereddict
  Building wheel for ruamel.ordereddict (setup.py) ... error
ERROR: Command errored out with exit status 1:
   command: /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/setup.py'"'"'; __file__='"'"'/private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-wheel-0GXQ8T
       cwd: /private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/
  Complete output (14 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.9-x86_64-2.7
  creating build/lib.macosx-10.9-x86_64-2.7/ruamel
  creating build/lib.macosx-10.9-x86_64-2.7/ruamel/ordereddict
  copying ./__init__.py -> build/lib.macosx-10.9-x86_64-2.7/ruamel/ordereddict
  running build_ext
  building '_ordereddict' extension
  creating build/temp.macosx-10.9-x86_64-2.7
  gcc -fno-strict-aliasing -fno-common -dynamic -arch x86_64 -g -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c ordereddict.c -o build/temp.macosx-10.9-x86_64-2.7/ordereddict.o
  xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for ruamel.ordereddict
  Running setup.py clean for ruamel.ordereddict
Failed to build ruamel.ordereddict
Installing collected packages: ruamel.ordereddict, ruamel.yaml, openshift
    Running setup.py install for ruamel.ordereddict ... error
    ERROR: Command errored out with exit status 1:
     command: /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/setup.py'"'"'; __file__='"'"'/private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-record-fjCzvl/install-record.txt --single-version-externally-managed --compile --install-headers /Library/Frameworks/Python.framework/Versions/2.7/include/python2.7/ruamel.ordereddict
         cwd: /private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/
    Complete output (14 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.macosx-10.9-x86_64-2.7
    creating build/lib.macosx-10.9-x86_64-2.7/ruamel
    creating build/lib.macosx-10.9-x86_64-2.7/ruamel/ordereddict
    copying ./__init__.py -> build/lib.macosx-10.9-x86_64-2.7/ruamel/ordereddict
    running build_ext
    building '_ordereddict' extension
    creating build/temp.macosx-10.9-x86_64-2.7
    gcc -fno-strict-aliasing -fno-common -dynamic -arch x86_64 -g -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c ordereddict.c -o build/temp.macosx-10.9-x86_64-2.7/ordereddict.o
    xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
    error: command 'gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/setup.py'"'"'; __file__='"'"'/private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-install-XOVWaq/ruamel-ordereddict/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/g8/q7jbx_f13fg2_2fb2dvhs3jc0000gn/T/pip-record-fjCzvl/install-record.txt --single-version-externally-managed --compile --install-headers /Library/Frameworks/Python.framework/Versions/2.7/include/python2.7/ruamel.ordereddict Check the logs for full command output.

geerlingguy, 2021-01-10:
It looks like the problem is this:

```
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
```

You may need to install the Xcode Command Line Tools on your Mac.

garylefebvreg, 2021-01-10:
Thank you, Jeff. I have installed Xcode Command Line Tools and have now a new error. with runnint the Ansible command
ansible-galaxy install -r requirements.yml --force

Garys-MacBook-Air:raspberry-pi-dramble-master garylefebvre$ ansible-galaxy install -r requirements.yml --force --ignore-errors
/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/ansible/parsing/vault/__init__.py:44: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Starting galaxy role install process
[WARNING]: - geerlingguy.security was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
- changing role geerlingguy.swap from 1.0.2 to unspecified
[WARNING]: - geerlingguy.swap was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
- changing role geerlingguy.nfs from 2.0.0 to unspecified
[WARNING]: - geerlingguy.nfs was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
- changing role geerlingguy.pip from 2.0.0 to unspecified
[WARNING]: - geerlingguy.pip was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
- changing role geerlingguy.docker_arm from 4.0.0 to unspecified
[WARNING]: - geerlingguy.docker_arm was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
- changing role geerlingguy.kubernetes from 5.0.1 to unspecified
[WARNING]: - geerlingguy.kubernetes was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
- changing role geerlingguy.k8s_manifests from 1.2.0 to unspecified
[WARNING]: - geerlingguy.k8s_manifests was NOT installed successfully: Unknown error when attempting to call Galaxy at
'https://galaxy.ansible.com/api/': <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>

garylefebvreg, 2021-01-17:
Sir, I was able to switch OS to an Ubuntu VM. I managed to run all scripts and the last was running to the very end and then errored out..

kind regards!

geerlingguy, 2021-01-26:
Excellent!

Please open up any specific issues you encounter elsewhere.

closed 2021-01-26 by geerlingguy