geerlingguy/raspberry-pi-dramble #201: Apply metrics-server manifest to the cluster fails

Flattermann78, 2021-03-15:
The task "Apply metrics-server manifest to the cluster" fails with the following message. I'm able to run the yml as root directly on the host ...

TASK [Apply metrics-server manifest to the cluster.] ******************************************************************************************************************************************************
task path: /home/pi/raspberry-pi-dramble/tasks/k8s-services.yml:10
Using module file /usr/lib/python3/dist-packages/ansible/modules/clustering/k8s/k8s.py
<10.0.100.61> ESTABLISH SSH CONNECTION FOR USER: pi
<10.0.100.61> SSH: EXEC ssh -C -o ControlMaster=auto -o ControlPersist=60s -o KbdInteractiveAuthentication=no -o PreferredAuthentications=gssapi-with-mic,gssapi-keyex,hostbased,publickey -o PasswordAuthentication=no -o User=pi -o ConnectTimeout=10 -o ControlPath=/tmp/ansible-ssh-%h-%p-%r 10.0.100.61 '/bin/sh -c '"'"'sudo -H -S -n -u root /bin/sh -c '"'"'"'"'"'"'"'"'echo BECOME-SUCCESS-wxwuxpteccmvhyzdsouvqgdwqmrtfjez; /usr/bin/python'"'"'"'"'"'"'"'"' && sleep 0'"'"''
Escalation succeeded
<10.0.100.61> (1, b'', b'Traceback (most recent call last):\n  File "<stdin>", line 113, in <module>\n  File "<stdin>", line 105, in _ansiballz_main\n  File "<stdin>", line 48, in invoke_module\n  File "/tmp/ansible_k8s_payload_7eRJbk/__main__.py", line 171, in <module>\n  File "/tmp/ansible_k8s_payload_7eRJbk/__main__.py", line 167, in main\n  File "/tmp/ansible_k8s_payload_7eRJbk/ansible_k8s_payload.zip/ansible/module_utils/k8s/raw.py", line 89, in execute_module\n  File "/tmp/ansible_k8s_payload_7eRJbk/ansible_k8s_payload.zip/ansible/module_utils/k8s/common.py", line 174, in get_api_client\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/client.py", line 71, in __init__\n    self.__discoverer = discoverer(self, cache_file)\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 259, in __init__\n    Discoverer.__init__(self, client, cache_file)\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 31, in __init__\n    self.__init_cache()\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 78, in __init_cache\n    self._load_server_info()\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 158, in _load_server_info\n    \'kubernetes\': self.client.request(\'get\', \'/version\', serializer=just_json)\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/client.py", line 42, in inner\n    resp = func(self, *args, **kwargs)\n  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/client.py", line 247, in request\n    _return_http_data_only=params.get(\'_return_http_data_only\', True)\n  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/api_client.py", line 353, in call_api\n    _preload_content, _request_timeout, _host)\n  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/api_client.py", line 184, in __call_api\n    _request_timeout=_request_timeout)\n  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/api_client.py", line 377, in request\n    headers=headers)\n  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/rest.py", line 243, in GET\n    query_params=query_params)\n  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/rest.py", line 216, in request\n    headers=headers)\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/request.py", line 75, in request\n    method, url, fields=fields, headers=headers, **urlopen_kw\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/request.py", line 96, in request_encode_url\n    return self.urlopen(method, url, **extra_kw)\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/poolmanager.py", line 375, in urlopen\n    response = conn.urlopen(method, u.request_uri, **kw)\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 796, in urlopen\n    **response_kw\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 796, in urlopen\n    **response_kw\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 796, in urlopen\n    **response_kw\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 756, in urlopen\n    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]\n  File "/usr/local/lib/python2.7/dist-packages/urllib3/util/retry.py", line 573, in increment\n    raise MaxRetryError(_pool, url, error or ResponseError(cause))\nurllib3.exceptions.MaxRetryError: HTTPConnectionPool(host=\'localhost\', port=80): Max retries exceeded with url: /version (Caused by NewConnectionError(\'<urllib3.connection.HTTPConnection object at 0x7fa9e2a250>: Failed to establish a new connection: [Errno 111] Connection refused\',))\n')
<10.0.100.61> Failed to connect to the host via ssh: Traceback (most recent call last):
  File "<stdin>", line 113, in <module>
  File "<stdin>", line 105, in _ansiballz_main
  File "<stdin>", line 48, in invoke_module
  File "/tmp/ansible_k8s_payload_7eRJbk/__main__.py", line 171, in <module>
  File "/tmp/ansible_k8s_payload_7eRJbk/__main__.py", line 167, in main
  File "/tmp/ansible_k8s_payload_7eRJbk/ansible_k8s_payload.zip/ansible/module_utils/k8s/raw.py", line 89, in execute_module
  File "/tmp/ansible_k8s_payload_7eRJbk/ansible_k8s_payload.zip/ansible/module_utils/k8s/common.py", line 174, in get_api_client
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/client.py", line 71, in __init__
    self.__discoverer = discoverer(self, cache_file)
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 259, in __init__
    Discoverer.__init__(self, client, cache_file)
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 31, in __init__
    self.__init_cache()
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 78, in __init_cache
    self._load_server_info()
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/discovery.py", line 158, in _load_server_info
    'kubernetes': self.client.request('get', '/version', serializer=just_json)
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/client.py", line 42, in inner
    resp = func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openshift/dynamic/client.py", line 247, in request
    _return_http_data_only=params.get('_return_http_data_only', True)
  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/api_client.py", line 353, in call_api
    _preload_content, _request_timeout, _host)
  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/api_client.py", line 184, in __call_api
    _request_timeout=_request_timeout)
  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/api_client.py", line 377, in request
    headers=headers)
  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/rest.py", line 243, in GET
    query_params=query_params)
  File "/usr/local/lib/python2.7/dist-packages/kubernetes/client/rest.py", line 216, in request
    headers=headers)
  File "/usr/local/lib/python2.7/dist-packages/urllib3/request.py", line 75, in request
    method, url, fields=fields, headers=headers, **urlopen_kw
  File "/usr/local/lib/python2.7/dist-packages/urllib3/request.py", line 96, in request_encode_url
    return self.urlopen(method, url, **extra_kw)
  File "/usr/local/lib/python2.7/dist-packages/urllib3/poolmanager.py", line 375, in urlopen
    response = conn.urlopen(method, u.request_uri, **kw)
  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 796, in urlopen
    **response_kw
  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 796, in urlopen
    **response_kw
  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 796, in urlopen
    **response_kw
  File "/usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 756, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/usr/local/lib/python2.7/dist-packages/urllib3/util/retry.py", line 573, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded with url: /version (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa9e2a250>: Failed to establish a  [... cut ...]
      File "/usr/local/lib/python2.7/dist-packages/urllib3/util/retry.py", line 573, in increment
        raise MaxRetryError(_pool, url, error or ResponseError(cause))
    urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded with url: /version (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa9e2a250>: Failed to establish a new connection: [Errno 111] Connection refused',))
  module_stdout: ''
  msg: |-
    MODULE FAILURE
    See stdout/stderr for the exact error
  rc: 1

flakeman2, 2021-03-19:
I have the same problem.  Should this be running under python2.7 or python3?

flakeman2, 2021-03-20:
FYI - I tried switching to python3 (ansible all -i inventory -a "ln -sf /usr/bin/python3 /usr/bin/python" -b) but I'm still getting the same error:

`TASK [Apply metrics-server manifest to the cluster.] *********************************************************************************************************************************************************
fatal: [kube1 -> 192.168.0.61]: FAILED! => changed=false 
  module_stderr: |-
    /usr/lib/python3/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.26.3) or chardet (3.0.4) doesn't match a supported version!
      RequestsDependencyWarning)
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 170, in _new_conn
        (self._dns_host, self.port), self.timeout, **extra_kw
      File "/usr/local/lib/python3.7/dist-packages/urllib3/util/connection.py", line 96, in create_connection
        raise err
      File "/usr/local/lib/python3.7/dist-packages/urllib3/util/connection.py", line 86, in create_connection
        sock.connect(sa)
    ConnectionRefusedError: [Errno 111] Connection refused
  
    During handling of the above exception, another exception occurred:
  
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connectionpool.py", line 706, in urlopen
        chunked=chunked,
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connectionpool.py", line 394, in _make_request
        conn.request(method, url, **httplib_request_kw)
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 234, in request
        super(HTTPConnection, self).request(method, url, body=body, headers=headers)
      File "/usr/lib/python3.7/http/client.py", line 1244, in request
        self._send_request(method, url, body, headers, encode_chunked)
      File "/usr/lib/python3.7/http/client.py", line 1290, in _send_request
        self.endheaders(body, encode_chunked=encode_chunked)
      File "/usr/lib/python3.7/http/client.py", line 1239, in endheaders
        self._send_output(message_body, encode_chunked=encode_chunked)
      File "/usr/lib/python3.7/http/client.py", line 1026, in _send_output
        self.send(msg)
      File "/usr/lib/python3.7/http/client.py", line 966, in send
        self.connect()
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 200, in connect
        conn = self._new_conn()
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 182, in _new_conn
        self, "Failed to establish a new connection: %s" % e
    urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7f8da1fc88>: Failed to establish a new connection: [Errno 111] Connection refused
  
    During handling of the above exception, another exception occurred:
  
    Traceback (most recent call last):
      File "<stdin>", line 102, in <module>
      File "<stdin>", line 94, in _ansiballz_main
      File "<stdin>", line 40, in invoke_module
      File "/usr/lib/python3.7/runpy.py", line 205, in run_module
        return _run_module_code(code, init_globals, run_name, mod_spec)
      File "/usr/lib/python3.7/runpy.py", line 96, in _run_module_code
        mod_name, mod_spec, pkg_name, script_name)
      File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
        exec(code, run_globals)
      File "/tmp/ansible_k8s_payload_kimn0u3s/ansible_k8s_payload.zip/ansible/modules/clustering/k8s/k8s.py", line 279, in <module>
      File "/tmp/ansible_k8s_payload_kimn0u3s/ansible_k8s_payload.zip/ansible/modules/clustering/k8s/k8s.py", line 275, in main
      File "/tmp/ansible_k8s_payload_kimn0u3s/ansible_k8s_payload.zip/ansible/module_utils/k8s/raw.py", line 171, in execute_module
      File "/tmp/ansible_k8s_payload_kimn0u3s/ansible_k8s_payload.zip/ansible/module_utils/k8s/common.py", line 200, in get_api_client
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/client.py", line 71, in __init__
        self.__discoverer = discoverer(self, cache_file)
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/discovery.py", line 259, in __init__
        Discoverer.__init__(self, client, cache_file)
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/discovery.py", line 31, in __init__
        self.__init_cache()
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/discovery.py", line 78, in __init_cache
        self._load_server_info()
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/discovery.py", line 158, in _load_server_info
        'kubernetes': self.client.request('get', '/version', serializer=just_json)
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/client.py", line 42, in inner
        resp = func(self, *args, **kwargs)
      File "/usr/local/lib/python3.7/dist-packages/openshift/dynamic/client.py", line 247, in request
[... 9 lines cut ...]
      File "/usr/local/lib/python3.7/dist-packages/kubernetes/client/rest.py", line 216, in request
        headers=headers)
      File "/usr/local/lib/python3.7/dist-packages/urllib3/request.py", line 75, in request
        method, url, fields=fields, headers=headers, **urlopen_kw
      File "/usr/local/lib/python3.7/dist-packages/urllib3/request.py", line 96, in request_encode_url
        return self.urlopen(method, url, **extra_kw)
[... cut ...]
      File "/usr/local/lib/python3.7/dist-packages/urllib3/util/retry.py", line 573, in increment
        raise MaxRetryError(_pool, url, error or ResponseError(cause))
    urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded with url: /version (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8da1fc88>: Failed to establish a new connection: [Errno 111] Connection refused'))
  module_stdout: ''
  msg: |-
    MODULE FAILURE
    See stdout/stderr for the exact error
  rc: 1
`

Flattermann78, 2021-03-23:
It seems to the untrained eye that the kubectl command is not issued as a root user and fails accordingly

Flattermann78, 2021-03-29:
I used the wrong ansible version. Replaced by the most recent version from pip solved the issue.

closed 2021-03-29 by Flattermann78

flakeman2, 2021-03-29:
@Flattermann78 what version of ansible and pip did you end of using?