geerlingguy/raspberry-pi-dramble #202: Question: K8S and HA - what is the impact on Rpi storage?

cmonty14, 2021-03-22:
Hello,

what are your thoughts about K8S and HA?
Rancher is "promoting" this with an [External DB](https://rancher.com/docs/k3s/latest/en/installation/ha/) and [Embedded DB](https://rancher.com/docs/k3s/latest/en/installation/ha-embedded/).

The Rpi dramble would allow me to setup 3 master nodes + x worker nodes.
But what is the impact on the most vulnerable HW of an Rpi: SD card?

I could image that an Embedded DB will generate many write IO, probably too many for an SD card.

Do you have any experience with an Embedded DB deployment of K8S?

Regards
Thomas

geerlingguy, 2021-03-22:
See related: https://github.com/geerlingguy/turing-pi-cluster/issues/33

I would suggest not using onboard eMMC or microSD for embedded databases in a production environment. Rather, use an external USB 2.0 or USB 3.0 SSD (preferred) or HDD.

flakeman2, 2021-03-23:
I added SSD's to my cluster.  Obviously it increases the cost but you get some nice performance gains :)
[image]

cmonty14, 2021-03-23:
> See related: [geerlingguy/turing-pi-cluster#33](https://github.com/geerlingguy/turing-pi-cluster/issues/33)
> 
> I would suggest not using onboard eMMC or microSD for embedded databases in a production environment. Rather, use an external USB 2.0 or USB 3.0 SSD (preferred) or HDD.

Do you recommend to avoid using eMMC in general incl. eMMC modules (addon)?

geerlingguy, 2021-03-23:
eMMC is going to be better than microSD, but still only about half the improvement you'd get by using SSD or something else external to the Pi.

closed 2021-07-22 by stale[bot]