geerlingguy/raspberry-pi-dramble #206: Failed to fetch package

adithyak04, 2021-05-24:
I am currently in `testing/vagrant` folder and I have first executed `vagrant up` command followed by `vagrant provision` command. After executing the provision command, I am getting the following error:

`
fatal: [kube1]: FAILED! => changed=false 
  cache_update_time: 1621798153
  cache_updated: false
  msg: |-
    '/usr/bin/apt-get -y -o "Dpkg::Options::=--force-confdef" -o "Dpkg::Options::=--force-confold"      install 'python3-pip'' failed: E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
  rc: 100
  stderr: |-
    E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
  stderr_lines: <omitted>
  stdout: ''
  stdout_lines: <omitted>
fatal: [kube2]: FAILED! => changed=false 
  cache_update_time: 1621798256
  cache_updated: false
  msg: |-
    '/usr/bin/apt-get -y -o "Dpkg::Options::=--force-confdef" -o "Dpkg::Options::=--force-confold"      install 'python3-pip'' failed: E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
  rc: 100
  stderr: |-
    E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
  stderr_lines: <omitted>
  stdout: ''
  stdout_lines: <omitted>
fatal: [kube4]: FAILED! => changed=false 
  cache_update_time: 1604516116
  cache_updated: false
  msg: |-
    '/usr/bin/apt-get -y -o "Dpkg::Options::=--force-confdef" -o "Dpkg::Options::=--force-confold"      install 'python3-pip'' failed: E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/l/linux/linux-libc-dev_4.19.152-1_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Failed to fetch http://httpredir.debian.org/debian/pool/main/p/python3.7/libpython3.7_3.7.3-2+deb10u2_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Failed to fetch http://httpredir.debian.org/debian/pool/main/p/python3.7/libpython3.7-dev_3.7.3-2+deb10u2_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Failed to fetch http://httpredir.debian.org/debian/pool/main/p/python3.7/python3.7-dev_3.7.3-2+deb10u2_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
  rc: 100
  stderr: |-
    E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/l/linux/linux-libc-dev_4.19.152-1_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Failed to fetch http://httpredir.debian.org/debian/pool/main/p/python3.7/libpython3.7_3.7.3-2+deb10u2_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Failed to fetch http://httpredir.debian.org/debian/pool/main/p/python3.7/libpython3.7-dev_3.7.3-2+deb10u2_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Failed to fetch http://httpredir.debian.org/debian/pool/main/p/python3.7/python3.7-dev_3.7.3-2+deb10u2_amd64.deb  404  Not Found [IP: 199.232.46.132 80]
    E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
  stderr_lines: <omitted>
  stdout: |-
    Reading package lists...
    Building dependency tree...
    Reading state information...
    The following additional packages will be installed:
      build-essential cpp cpp-8 dh-python dpkg-dev fakeroot g++ g++-8 gcc gcc-8
      libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
      libasan5 libatomic1 libc-dev-bin libc6-dev libcc1-0 libdpkg-perl
      libexpat1-dev libfakeroot libfile-fcntllock-perl libgcc-8-dev libgomp1
      libisl19 libitm1 liblsan0 libmpc3 libmpfr6 libmpx2 libpython3-dev
      libpython3.7 libpython3.7-dev libquadmath0 libstdc++-8-dev libtsan0
      libubsan1 linux-libc-dev manpages-dev python3-asn1crypto
      python3-cffi-backend python3-crypto python3-cryptography python3-dev
      python3-distutils python3-entrypoints python3-keyring python3-keyrings.alt
      python3-lib2to3 python3-secretstorage python3-setuptools python3-wheel
      python3-xdg python3.7-dev
    Suggested packages:
      cpp-doc gcc-8-locales debian-keyring g++-multilib g++-8-multilib gcc-8-doc
      libstdc++6-8-dbg gcc-multilib autoconf automake libtool flex bison gdb
      gcc-doc gcc-8-multilib libgcc1-dbg libgomp1-dbg libitm1-dbg libatomic1-dbg
      libasan5-dbg liblsan0-dbg libtsan0-dbg libubsan1-dbg libmpx2-dbg
      libquadmath0-dbg glibc-doc bzr libstdc++-8-doc python-crypto-doc
      python-cryptography-doc python3-cryptography-vectors gnome-keyring
      libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-secretstorage-doc
      python-setuptools-doc
    The following NEW packages will be installed:
      build-essential cpp cpp-8 dh-python dpkg-dev fakeroot g++ g++-8 gcc gcc-8
      libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
      libasan5 libatomic1 libc-dev-bin libc6-dev libcc1-0 libdpkg-perl
      libexpat1-dev libfakeroot libfile-fcntllock-perl libgcc-8-dev libgomp1
      libisl19 libitm1 liblsan0 libmpc3 libmpfr6 libmpx2 libpython3-dev
      libpython3.7 libpython3.7-dev libquadmath0 libstdc++-8-dev libtsan0
      libubsan1 linux-libc-dev manpages-dev python3-asn1crypto
      python3-cffi-backend python3-crypto python3-cryptography python3-dev
      python3-distutils python3-entrypoints python3-keyring python3-keyrings.alt
      python3-lib2to3 python3-pip python3-secretstorage python3-setuptools
      python3-wheel python3-xdg python3.7-dev
    0 upgraded, 55 newly installed, 0 to remove and 0 not upgraded.
    Need to get 51.8 MB/96.8 MB of archives.
    After this operation, 254 MB of additional disk space will be used.
    Err:1 http://security.debian.org/debian-security buster/updates/main amd64 linux-libc-dev amd64 4.19.152-1
      404  Not Found [IP: 199.232.46.132 80]
    Err:2 http://httpredir.debian.org/debian buster/main amd64 libpython3.7 amd64 3.7.3-2+deb10u2
      404  Not Found [IP: 199.232.46.132 80]
    Err:3 http://httpredir.debian.org/debian buster/main amd64 libpython3.7-dev amd64 3.7.3-2+deb10u2
      404  Not Found [IP: 199.232.46.132 80]
    Err:4 http://httpredir.debian.org/debian buster/main amd64 python3.7-dev amd64 3.7.3-2+deb10u2
      404  Not Found [IP: 199.232.46.132 80]
  stdout_lines: <omitted>`

geerlingguy, 2021-05-24:
@adithyak04 - Can you log into the machine and manually run a `sudo apt-get update` and see if that succeeds?

adithyak04, 2021-05-24:
>  `sudo apt-get update` worked for me. Now, one more error has come up:
`fatal: [kube3]: FAILED! => changed=false
   msg: No package matching 'containerd.io' is available`

geerlingguy, 2021-05-24:
Hmm... are you running the entire playbook on the cluster?

If so, I wonder if that same error is occurring on any of the other instances or just `kube3`?

adithyak04, 2021-05-24:
The error is occurring on all the instances

closed 2021-09-22 by stale[bot]