geerlingguy/raspberry-pi-dramble #209: Replace Blinkstick with Adafruit Neo Trinkey

jpconstantineau, 2021-08-18:
Would it be possible to change the expensive Blinkstick with the cheaper [Adafruit Neo Trinkey](https://www.adafruit.com/product/4870)?
It's about 7$ compared to €16.50 (about 20$), not including shipping.

The Trinkey runs CircuitPython too.  As such, it would be possible to add basic serial port functionality and program it to respond to serial commands from the computer to make it run similar to a Blinkstick and even extend it to do more interesting things (it has 2 touch keys).

geerlingguy, 2021-08-19:
That would definitely be doable, I just haven't tested anything with a Trinkey :)

geerlingguy, 2021-08-19:
Would be neat to have different modes, like switch between blinking when there's network activity vs showing red/green for kubectl status.

jpconstantineau, 2021-08-24:
The trinkeys have 4 RGB LEDs.  Not sure yet from a distance how much they blend in or if one can differentiate between them. 
This means that multiple statuses could be displayed.

Just for reference when I receive my Trinkeys...
There is example code to set this up on the Trinkey side: 
https://github.com/Neradoc/circuitpython-sample-scripts/tree/main/serial

I'll need help on the server side.

jpconstantineau, 2021-12-26:
Hi Jeff.  
Found out this [great repo](https://github.com/daveparker/neotrinkey) that will help kick things off on this. 
Specifically the serial examples.