geerlingguy/raspberry-pi-dramble #211: TASK [geerlingguy.containerd : Ensure containerd is installed.]

jdalke, 2021-08-24:
Running the Ansible Playbook I received the following error - 

TASK [geerlingguy.containerd : Ensure containerd is installed.] ***********************************************************************************************************************************************
fatal: [kube2]: FAILED! => changed=false
  msg: No package matching 'containerd.io' is available
fatal: [kube3]: FAILED! => changed=false
  msg: No package matching 'containerd.io' is available
fatal: [kube4]: FAILED! => changed=false
  msg: No package matching 'containerd.io' is available  

Ansible is add the docker repository to the /etc/apt/sources.list.d/ directory but won't install.

jdalke, 2021-08-24:
The issue is I did not pay attention to the instructions on the Setting up the PI's in the readme where it states:   

_"It is recommended you use the 64-bit version of Raspberry Pi OS, as some of this project's dependencies require it (and may not install on 32-bit Pi OS)."_ 

The resolution was to re-image the SD cards with Raspi-Lite-ARM64 version. Once you use a 64 bit OS the playbook works! This is not a recommendation but a MUST for newbies.

jdalke, 2021-08-24:
I posted this issue as I spent allot of time figuring this out and hopefully this will help other people :)

geerlingguy, 2021-08-24:
@jdalke - Ah, good catch! It's been over a year now since I switched to only running the 64-bit Pi OS, it's been very stable and so many more things 'just work' there :)

jdalke, 2021-08-24:
Enjoying this project allot as it's forcing me to learn allot of new technologies. Thanks for posting and maintaining it!

closed 2021-08-24 by jdalke