geerlingguy/raspberry-pi-dramble #213: raspberry-pi-dramble/tasks/k8s-services.yml : urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded

jdalke, 2021-08-25:
Why running the ansible playbook I received the following error: 

TASK [Apply metrics-server manifest to the cluster.] **********************************************************************************************************************************************************
task path: /home/jdalke/raspberry-pi-dramble/tasks/k8s-services.yml:10
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded with url: /version (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fb5a1ff98>: Failed to establish a new connection: [Errno 111] Connection refused'))
fatal: [kube1 -> 192.168.1.221]: FAILED! => changed=false
  module_stderr: |-
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 170, in _new_conn
        (self._dns_host, self.port), self.timeout, **extra_kw
      File "/usr/local/lib/python3.7/dist-packages/urllib3/util/connection.py", line 96, in create_connection
        raise err
      File "/usr/local/lib/python3.7/dist-packages/urllib3/util/connection.py", line 86, in create_connection
        sock.connect(sa)
    ConnectionRefusedError: [Errno 111] Connection refused

This is unique as I had to use a 192.1.168.x network due to my work VPN using the recommended 10.1.x.x. 
Initial though is I missed something in the network setup as direction on altering the solution away from the 10.x network don't exist!

floatingman, 2021-09-08:
One thing I changed was in [setup/networking/templates/hosts.j2](https://github.com/geerlingguy/raspberry-pi-dramble/blob/master/setup/networking/templates/hosts.j2)

and changed line 12 to be 
`192.168.0.{{ 49 + loop.index }} kube{{ loop.index }} kube{{ loop.index }}.pidramble.com`

That didn't quite fix my problem, but I think I was having an issue because I was trying to run Ansible in WSL on Windows.

When I switched to my linux machine, I was able to succsfully run the playbook.

jdalke, 2021-09-10:
**Following is the IP addresses I decided to use and coded in my setup/networking/inventory file :** 
[dramble]
192.168.1.221
192.168.1.222
192.168.1.223
192.168.1.224

[dramble:vars]
ansible_ssh_user=pi

**Below it the hosts.j2 settings** 

127.0.0.1 localhost
::1   localhost ip6-localhost ip6-loopback
fe00::0   ip6-localnet
ff00::0   ip6-mcastprefix
ff02::1   ip6-allnodes
ff02::2   ip6-allrouters

127.0.1.1 {{ dramble_hostname }}

{# Add an entry for each host in the cluster, using the loop index to map IPs #}
{% for host in groups['dramble'] %}
192.168.1.{{ 200 + loop.index }} kube{{ loop.index }} kube{{ loop.index }}.pidramble.com
{% endfor %}

**I am assuming this is correct as the network setup ansible worked.**

jdalke, 2021-09-10:
**Also when I go to kube1 thru 4 and look at the /etc/hosts it looks correct. Here is what is in kube1 hosts file**

127.0.0.1 localhost
::1   localhost ip6-localhost ip6-loopback
fe00::0   ip6-localnet
ff00::0   ip6-mcastprefix
ff02::1   ip6-allnodes
ff02::2   ip6-allrouters

127.0.1.1 kube1.pidramble.com

192.168.1.201 kube1 kube1.pidramble.com
192.168.1.202 kube2 kube2.pidramble.com
192.168.1.203 kube3 kube3.pidramble.com
192.168.1.204 kube4 kube4.pidramble.com
192.168.1.223  registry.pidramble.test

floatingman, 2021-09-10:
Sure it all depends on your network configuration. Just make the starting loop index one less than what you want to start with.

This was the only place where 10.0.100.X netmask is hardcoded.

jdalke, 2021-09-10:
The hosts files are correct on the PI's so I am assuming it worked correct. Since I started at 201 I updated hosts.j2 with the following

192.168.1.{{ **200** + loop.index }} kube{{ loop.index }} kube{{ loop.index }}.pidramble.com

jdalke, 2021-09-10:
I don't believe networking is the issue. Will post the ansible playbook response with the -vvv option

jdalke, 2021-09-10:
When running with the -vvv option it is very verbose so I have attached a file with the results. I think the error may be: 

fatal: [kube1 -> 192.168.1.221]: FAILED! => changed=false
  module_stderr: |-
    /usr/lib/python3/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.26.6) or chardet (3.0.4) doesn't match a supported version!
[ansible playbook error.txt](https://github.com/geerlingguy/raspberry-pi-dramble/files/7144523/ansible.playbook.error.txt)

      RequestsDependencyWarning)

jdalke, 2021-09-10:
Searching the above error resulted in Issue 201 where the resolution was they were running the wrong ansible version. Unfortunately the issue did not state what version was causing the issue and what version resolved it. 

I'm running  the following: 
ansible 2.9.6
  config file = /home/jdalke/raspberry-pi-dramble/ansible.cfg
  configured module search path = ['/home/jdalke/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/lib/python3/dist-packages/ansible
  executable location = /usr/bin/ansible
  python version = 3.8.10 (default, Jun  2 2021, 10:49:15) [GCC 9.4.0]

floatingman, 2021-09-10:
I'm running 2.11.4.  What environment are you using? (Windows, Ubuntu, Mac)

jdalke, 2021-09-10:
Running on Ubuntu 20.04

jdalke, 2021-09-10:
Sorry - on a dedicated linux box

floatingman, 2021-09-10:
You could look into installing a later version of Ansible directly from pip.

https://docs.ansible.com/ansible/latest/installation_guide/intro_installation.html#upgrading-from-ansible-3-or-ansible-core-2-10

The only other thing I've changed is to a more recent version of Kubernetes in vars/main.yml

```
kubernetes_version: '1.22'
kubernetes_packages:
  - name: kubelet=1.22.1-00
    state: present
  - name: kubectl=1.22.1-00
    state: present
  - name: kubeadm=1.22.1-00
    state: present
  - name: kubernetes-cni
    state: present
```

jdalke, 2021-09-10:
PIP3 install did not work on Unbuntu as after the install the ansible --version gave me nothing. Installed using APT: 
sudo add-apt-repository --yes --update ppa:ansible/ansible-2.10

Ansible version -  
ansible [core 2.11.4]
  config file = /home/jdalke/raspberry-pi-dramble/ansible.cfg
  configured module search path = ['/home/jdalke/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /home/jdalke/.local/lib/python3.8/site-packages/ansible
  ansible collection location = /home/jdalke/.ansible/collections:/usr/share/ansible/collections
  executable location = /usr/bin/ansible
  python version = 3.8.10 (default, Jun  2 2021, 10:49:15) [GCC 9.4.0]
  jinja version = 2.10.1
  libyaml = True

Also update the vars/main.yml as directed and now getting the following error:  

statically imported: /home/jdalke/raspberry-pi-dramble/tasks/k8s-services.yml
redirecting (type: modules) ansible.builtin.k8s to kubernetes.core.k8s
ERROR! couldn't resolve module/action 'k8s'. This often indicates a misspelling, missing collection, or incorrect module path.

The error appears to be in '/home/jdalke/raspberry-pi-dramble/tasks/k8s-services.yml': line 10, column 3, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:
- name: Apply metrics-server manifest to the cluster.
  ^ here

floatingman, 2021-09-10:
Interesting. I wonder if you have to do an ansible-galaxy install -r
requirements.yml after changing the version?

The PIP3 error was probably due to the global Python site-packages
directory not being in your path. Did you install with pip3 using sudo?

I feel Ubuntu makes things difficult because it defaults to Python 2.7 vs
Python 3.

jdalke, 2021-09-10:
**Running ansible-galaxy install -r requirements.yml provides the following so assuming everything okay**
Starting galaxy role install process
[WARNING]: - geerlingguy.security (2.0.1) is already installed - use --force to change version to unspecified
[WARNING]: - geerlingguy.swap (1.0.2) is already installed - use --force to change version to unspecified
[WARNING]: - geerlingguy.nfs (2.0.0) is already installed - use --force to change version to unspecified
[WARNING]: - geerlingguy.pip (2.1.0) is already installed - use --force to change version to unspecified
[WARNING]: - geerlingguy.containerd (1.1.1) is already installed - use --force to change version to unspecified
[WARNING]: - geerlingguy.kubernetes (7.1.0) is already installed - use --force to change version to unspecified
[WARNING]: - geerlingguy.k8s_manifests (1.2.0) is already installed - use --force to change version to unspecified

Also I'm assuming I installed PIP3 under sudo as I can run PIP3 commands. 

What version of linux are you running?

floatingman, 2021-09-10:
I'm running Arch Linux, which is why I think I have a later version of
Ansible. Sorry that I can't be more helpful.

jdalke, 2021-09-13:
Downgraded to Ansible 2:10 and past first error and right back where I started: 

TASK [Apply metrics-server manifest to the cluster.] ***************************************************************************************************************
fatal: [kube1 -> kube1]: FAILED! => changed=false
  msg: 'Failed to get client due to HTTPConnectionPool(host=''localhost'', port=80): Max retries exceeded with url: /version (Caused by NewConnectionError(''<urllib3.connection.HTTPConnection object at 0x7f8ef8eef0>: Failed to establish a new connection: [Errno 111] Connection refused''))'

Very frustrating when these playbooks crap and your new to ansible. Looking at what the metrics-server is trying to do.

closed 2022-01-14 by stale[bot]

jtcleek, 2022-05-15:
@jdalke Did you ever get past this error? I am currently feeling the frustration you expressed.