geerlingguy/raspberry-pi-dramble #214: `nfs-client-provisioner` won't start on Ubuntu Server 64-bit

the-wondersmith, 2021-09-02:
I have no idea if it's pertinent at all, but my cluster is made up of 3 RPi 4 8GB units and I wanted to take advantage of that so I imaged them with the 64-bit Ubuntu server image available via Raspberry Pi Imager.

Anyway, long story short the `nfs-client-provisioner` image that dramble is currently configured to use refuses to start on my RPis, but (this image)[https://quay.io/repository/vbouchaud/nfs-client-provisioner?tag=latest] appears to be a drop-in replacement that should work on any arm-based system regardless of bittedness.

I'm just starting out learning k8s, so I don't feel qualified to render an opinion on whether or not the current image should be replaced. I figured I'd just let you know and trust you to know better than I do. 😅

geerlingguy, 2021-09-02:
I'd count it as a bug. I need to run through the playbook soon and make sure to update the things like this that need updating!

the-wondersmith, 2021-09-02:
@geerlingguy Well, I'm glad I was able to find a drop-in replacement. 😁