geerlingguy/raspberry-pi-dramble #221: iozone compile error

jaui, 2021-11-18:
running 
`$ sudo ./microsd-benchmarks.sh`
under the latest Raspberry Pi OS based on bullseye (debian 11)
raises an error during compilation of **iozone**
```
/usr/bin/ld: libbif.o:(.bss+0x0): multiple definition of `junk'; iozone_linux-arm.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make: *** [makefile:221: linux-arm] Error 1
```

The hint from wesite
https://github.com/phoronix-test-suite/test-profiles/issues/139
add a `CFLAGS=-fcommon` statement to make helped.

`$ make CFLAGS=-fcommon linux-arm `
runs  smootly.

Would you please add this fix to make of iozone.

geerlingguy, 2021-11-18:
@jaui - I've just been testing this on two different builds of Raspberry Pi OS Bullseye over the past week, and haven't had any trouble building—though that was on the `disk-benchmarks.sh` script. Maybe the iozone version is old and needs updating?

2sls, 2021-12-16:
I had same issue  - for me, updating the version from 489 to 492 in the microsd-benchmark.sh file fixed everything

geerlingguy, 2021-12-16:
Bumped the version in all three files to 492 :)

commit 049148c8 (2021-12-16): Fixes #221: Bump IOZone version in all benchmark scripts.
closed 2021-12-16 by geerlingguy