geerlingguy/raspberry-pi-dramble #224: All net.bridge tasks are being skipped

sdaynjax, 2022-03-30:
Looks like all the net_bridge tasks are being skipped.  I noticed in the playbook there is a conditional that shows the task will run if the Ansible dist is not Debian or the major version is less than 10.  This is what I have,  I installed Raspberry Pi OS Lite (64 bit) a port of Debian that is compatible for Pi 3/4/400. Is any of this my problem.  My nodes are not able to connect to the master and I think this is the cause.

ansible [core 2.12.3]
  config file = None
  configured module search path = ['/home/pi/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/local/lib/python3.9/dist-packages/ansible
  ansible collection location = /home/pi/.ansible/collections:/usr/share/ansible/collections
  executable location = /usr/local/bin/ansible
  python version = 3.9.2 (default, Feb 28 2021, 17:03:44) [GCC 10.2.1 20210110]
  jinja version = 3.1.1
  libyaml = True

sdaynjax, 2022-04-03:
I resolved this by changing major version is < 10 to major version is > 10.

closed 2022-04-03 by sdaynjax

geerlingguy, 2022-04-04:
@sdaynjax - Was that a change you needed to make to this project's code? If so I can work on fixing it in the project too.